furrtek/portapack-havoc #124: Problem compiling?

chudgoo, 2017-11-10:
On the latest build from today I am getting hun up here:

[ 35%] Building CXX object firmware/application/CMakeFiles/application.elf.dir/ui_jammer.cpp.obj
In file included from /Users/chud/temphavoc/portapack-havoc/firmware/application/ui_jammer.cpp:23:0:
/Users/chud/temphavoc/portapack-havoc/firmware/application/ui_jammer.hpp:234:86: error: array must be initialized with a brace-enclosed initializer
  std::array<RangeView*, 3> range_views { &view_range_a, &view_range_b, &view_range_c };
                                                                                      ^
/Users/chud/temphavoc/portapack-havoc/firmware/application/ui_jammer.hpp:234:86: error: too many initializers for 'std::array<ui::RangeView*, 3u>'
make[3]: *** [firmware/application/CMakeFiles/application.elf.dir/ui_jammer.cpp.obj] Error 1
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2

Does anyone know what might be causing this?

furrtek, 2017-11-10:
You're probably using an outdated version of the toolchain, please use a recent one (2017) from https://developer.arm.com/open-source/gnu-toolchain/gnu-arm-embedded-toolchain/downloads
Let me know if you have troubles getting it to run.

chudgoo, 2017-11-10:
$ ~/opt/gcc-arm-none-eabi-6-2017-q2-update/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

I'm running the latest version and it failed at the spot again. Hmmmm.....

furrtek, 2017-11-10:
That's strange, I have the exact same version and I'm not getting that error. I added double-braces in f128b9b0b7cf98e49dcabf7eebe0729378e2086a as suggested on the GCC bug report discussion. Let me know if that works for you.

chudgoo, 2017-11-10:
It made it a little further!  ;)

[ 37%] Building CXX object firmware/application/CMakeFiles/application.elf.dir/ui_navigation.cpp.obj
In file included from /Users/chud/temphavoc/portapack-havoc/firmware/application/ui_navigation.cpp:54:0:
/Users/chud/temphavoc/portapack-havoc/firmware/application/ui_scanner.hpp:117:44: error: array must be initialized with a brace-enclosed initializer
  std::array<Color, 240> spectrum_row = { 0 };
                                            ^
In file included from /Users/chud/temphavoc/portapack-havoc/firmware/application/ui_soundboard.hpp:31:0,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/application/ui_navigation.cpp:59:
/Users/chud/temphavoc/portapack-havoc/firmware/application/io_wave.hpp:97:36: error: invalid conversion from 'const char*' to 'char' [-fpermissive]
  char artist[12] { "PortaPack\0\0" };
                                    ^
make[3]: *** [firmware/application/CMakeFiles/application.elf.dir/ui_navigation.cpp.obj] Error 1
make[2]: *** [firmware/application/CMakeFiles/application.elf.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2

chudgoo, 2017-11-10:
WHat's weird is that it keep failing in different spots...

Scanning dependencies of target baseband_replay.elf
[ 28%] Building CXX object firmware/baseband/CMakeFiles/baseband_replay.elf.dir/proc_replay.cpp.obj
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_replay.cpp: In member function 'virtual void ReplayProcessor::execute(const buffer_c8_t&)':
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_replay.cpp:53:14: warning: unused variable result' [-Wunused-variable]
   const auto result = stream->read(iq_buffer.p, bytes_to_read);
              ^
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_replay.cpp:63:47: warning: narrowing conversion of '(((int)(((std::complex<short int>*)((ReplayProcessor*)this)->ReplayProcessor::iq_buffer.buffer_t<std::complex<short int> >::p) + ((sizetype)((i >> 3) * 4u)))->std::complex<short int>::real()) >> 8)' from 'int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]
    buffer.p[i] = { iq_buffer.p[i >> 3].real() >> 8, iq_buffer.p[i >> 3].imag() >> 8 };
                                               ^
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_replay.cpp:63:80: warning: narrowing conversion of '(((int)(((std::complex<short int>*)((ReplayProcessor*)this)->ReplayProcessor::iq_buffer.buffer_t<std::complex<short int> >::p) + ((sizetype)((i >> 3) * 4u)))->std::complex<short int>::imag()) >> 8)' from 'int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]
    buffer.p[i] = { iq_buffer.p[i >> 3].real() >> 8, iq_buffer.p[i >> 3].imag() >> 8 };
                                                                                ^
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_replay.cpp:42:9: warning: unused variable 'pos' [-Wunused-variable]
  size_t pos = 0;
         ^
[ 28%] Linking CXX executable baseband_replay.elf
[ 28%] Built target baseband_replay.elf
Scanning dependencies of target baseband_jammer.elf
[ 29%] Building CXX object firmware/baseband/CMakeFiles/baseband_jammer.elf.dir/proc_jammer.cpp.obj
[ 29%] Linking CXX executable baseband_jammer.elf
[ 29%] Built target baseband_jammer.elf
Scanning dependencies of target baseband_adsbrx.elf
[ 29%] Building CXX object firmware/baseband/CMakeFiles/baseband_adsbrx.elf.dir/proc_adsbrx.cpp.obj
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.cpp: In member function 'virtual void ADSBRXProcessor::execute(const buffer_c8_t&)':
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.cpp:38:7: warning: variable 'confidence' set but not used [-Wunused-but-set-variable]
  bool confidence, first_in_window, last_in_window;
       ^
In file included from /Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.cpp:23:0:
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.hpp: In instantiation of 'typename std::_MakeUniq<_Tp>::__single_object std::make_unique(_Args&& ...) [with _Tp = ADSBRXProcessor; _Args = {}; typename std::_MakeUniq<_Tp>::__single_object = std::unique_ptr<ADSBRXProcessor, std::default_delete<ADSBRXProcessor> >]':
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.cpp:155:71:   required from here
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.hpp:36:7: warning: 'ADSBRXProcessor::threshold' should be initialized in the member initialization list [-Weffc++]
 class ADSBRXProcessor : public BasebandProcessor {
       ^
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.hpp:36:7: warning: 'ADSBRXProcessor::threshold_low' should be initialized in the member initialization list [-Weffc++]
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.hpp:36:7: warning: 'ADSBRXProcessor::threshold_high' should be initialized in the member initialization list [-Weffc++]
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.hpp:36:7: warning: 'ADSBRXProcessor::null_count' should be initialized in the member initialization list [-Weffc++]
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.hpp:36:7: warning: 'ADSBRXProcessor::bit_count' should be initialized in the member initialization list [-Weffc++]
/Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.hpp:36:7: warning: 'ADSBRXProcessor::sample_count' should be initialized in the member initialization list [-Weffc++]
In file included from /usr/local/Cellar/gcc-arm-none-eabi/20160928/arm-none-eabi/include/c++/5.4.1/memory:81:0,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/common/fifo.hpp:28,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/common/message.hpp:43,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/baseband/channel_stats_collector.hpp:26,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/baseband/baseband_processor.hpp:27,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.hpp:26,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_adsbrx.cpp:23:
/usr/local/Cellar/gcc-arm-none-eabi/20160928/arm-none-eabi/include/c++/5.4.1/bits/unique_ptr.h:765:69: note: synthesized method 'ADSBRXProcessor::ADSBRXProcessor()' first required here 
     { return unique_ptr<_Tp>(new _Tp(std::forward<_Args>(__args)...)); }
                                                                     ^
[... cut ...]
In file included from /Users/chud/temphavoc/portapack-havoc/firmware/common/message.hpp:33:0,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/baseband/channel_stats_collector.hpp:26,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/baseband/baseband_processor.hpp:27,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_pocsag.hpp:28,
                 from /Users/chud/temphavoc/portapack-havoc/firmware/baseband/proc_pocsag.cpp:25:

ImDroided, 2017-11-10:
Man I would delete the portapack directory and do a git request

rm -rf portapack-havoc
git clone https://github.com/furrtek/portapack-havoc.git

chudgoo, 2017-11-10:
Yeah, I've been doing this repeatedly.  I was trying to get a script doing that fetches, compiles and installs all in one command, but lately the compile errors have halted that.

furrtek, 2017-11-10:
Do you have different versions of gcc installed ? Are you sure that the right version is being used by cmake ?

chudgoo, 2017-11-10:
Probably a little late to be mentioning this but I'm on a Mac (10.12.6).

$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.1.0 (clang-802.0.42)
Target: x86_64-apple-darwin16.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

$ cmake --version
cmake version 3.4.3

$ cmake ..
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/local/Cellar/gcc-arm-none-eabi/20160928/arm-none-eabi
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/chud/temphavoc/portapack-havoc/build

It looks like cmake is pointing to the wrong gcc-arm....

furrtek, 2017-11-11:
You can change that in /firmware/toolchain-arm-cortex-m.cmake
Instead of just "arm-none-eabi-gcc" and "arm-none-eabi-g++" at the beginning of the file, put the full path to the correct binaries.

chudgoo, 2017-11-11:
Huh!  Looks like that might have worked...

$ cmake ..
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/local/Caskroom/gcc-arm-embedded/6-2017-q2-update/gcc-arm-none-eabi-6-2017-q2-update/arm-none-eabi
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/chud/temphavoc/portapack-havoc/build

chudgoo, 2017-11-11:
Ok, now that it's working (!!!!).... here's my dumb bash script.

`#!/bin/bash
set -e
#Back up old install
#zip -r havoc-lastversion-backup.zip ~/portapack-havoc

#Clean up old shit
sudo rm -rf ~/temphavoc

#Get latest version of Portapack-Havoc and overwrite existing install
mkdir ~/temphavoc
cd ~/temphavoc
git clone https://github.com/furrtek/portapack-havoc
cd ./portapack-havoc
mkdir build
cd build
cmake ..
make firmware
cd ../firmware/

read -n 1 -s -r -p "Connect yo muthafuckin PortaPack and Press any key to continue"

sleep 1
dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
sleep 2
hackrf_spiflash -w portapack-h1-havoc.bin
sleep 5
cat << "EOF"
U N P L U G   Y O   M U T H A F U C K I N
 _   _   ___  _   _  _____ _____ 
| | | | / _ \| | | ||  _  /  __ \
| |_| |/ /_\ \ | | || | | | /  \/
|  _  ||  _  | | | || | | | |    
| | | || | | \ \_/ /\ \_/ / \__/\
\_| |_/\_| |_/\___/  \___/ \____/
       _,.
     ,` -.)                                           ^  ^  ^
    '( _/'-\\-.                                        \ | /
   /,|`--._,-^|            ,                            \|/
   \_| |`-._/||          ,'|---/\/\/\/\/\/\/\/\/\/\/\/\/***) ) )
     |  `-, / |         / X/                            /|\
     |     || |        / X/                            / | \
      `r-._||/   __   / X/                            V  V  V
  __,-<_     )`-/  `./ X/  
 '  \   `---'   \   / X/ 
     |           |./ X/  
     /           // X/     
 \_/' \         |/  /         
  |    |   _,^-'/  /              
  |    , ``  (\/  /_        
   \,.->._    \X-=/^         
   (  /   `-._//^`  
    `Y-.____(__}              
     |     {__)           
           ()`     
EOF

`

chudgoo, 2017-11-11:
It's sloppy and vulgar but after all THAT i got a little angry.  Here's a bash script (rename to .sh if you want). It SHOULD download, compile and install the latest Havoc. 

[new-havoc.txt](https://github.com/furrtek/portapack-havoc/files/1463131/new-havoc.txt)

gh0stshell, 2017-11-11:
After we identified the issue in a build that was causing hackrf_spiflash
to no longer work furrtek fixed it and I do not have use "dfu-util --device
1fc9:000c --download hackrf_one_usb.dfu --reset" before using
hackrf_spiflash. I just run hackrf_spiflash to flash the latest firmware.

gh0stshell, 2017-11-11:
Sorry I forgot to add my question of is the issue back again? Or do you do
that as part of your routine instead of choosing hackrf mode in havoc
before flashing?

furrtek, 2017-11-11:
hackrf_spiflash: update hackRF through the currently programmed firmware
dfu-util: update hackRF no matter what, even if bricked

chudgoo, 2017-11-11:
@furrtek Yeah, it's the same stuff, just with less interactivity needed.  Also, it brings derpy ASCII art and cussing.   Or did you mean something else?

chudgoo, 2017-11-12:
So In this case of a script like this, would it be advisable to only use hackrf_spiflash unless absolutely necessary?

gh0stshell, 2017-11-12:
no it will not hurt anything, he was just providing the reasons why to use
each one

fantuz, 2017-11-23:
Editing the firmware cmake spec file does the trick, KISS as @furrtek suggests.
But then I get:

> In file included from /home/max/portapack-havoc2/firmware/application/ui_navigation.cpp:54:0:
> /home/max/portapack-havoc2/firmware/application/ui_scanner.hpp:117:44: error: array must be initialized with a brace-enclosed initializer
>   std::array<Color, 240> spectrum_row = { 0 };
>                                             ^
> 

I still do not own a portapack but evaluating the ~200$ expense.

Cheers from GVA
73 de HB3YOE, Max

fantuz, 2017-11-23:
PS: using
 - /usr/local/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-g++ et
 - /usr/local/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc
from Ubuntu repos.

I very recently did the (not-so-traumatic) do-release-upgrade on my Ubuntu 14 LTS to 16 LTS, recovered GRC GQRX BLADERF HACKRF OSMO etc so I guess problem might be on my side, even if other firmwares still compile correctly...

Any hints ?

furrtek, 2017-11-27:
Updated the wiki with the new instructions since arm-gcc needs to be downloaded from ARM directly now.

closed 2020-01-03 by furrtek