furrtek/portapack-havoc #125: Suggestion: Adding Wifi Deauther

chimera234, 2017-11-11:
Hello,
Nice addon for exisiting Jammer would be a Deauther feature, details are available at:

https://github.com/spacehuhn/esp8266_deauther

The deauth attack will, if the connection is vulnerable, disconnect the devices from the network. Because the attack is running constantly, the devices will be disconnected again and again. Depending on the network, that can either block a connection or slow it down

Thanks

furrtek, 2017-11-11:
Hey,
The ESP8266 makes it seem simple but 802.11 is tough stuff, certainly too tough for the Portapack all alone.

Misfitsman805, 2017-11-17:
Could you not just use the jammer feature and inject tons of noise onto the wifi 2.4ghz band?

closed 2018-02-05 by chimera234