furrtek/portapack-havoc #126: Feature request: Allow Amp toggle for ADS-B Rx

unixpunk, 2017-11-13:
For some reason my HackRf (with/without PortaPack) sees no ADS-B signals when the Amp is enabled regardless of the combination of gain settings used.  Only until I disabled the Amp in the ui_adsb_rx.cpp was I able to get it to work and see planes on the list.  Not sure if anyone else has this issue or if something is up with my hw, but the ability to toggle would help out, even in cases of nearby high-powered transmitters that, without a BP/Notch filter, will always overload the internal amp.  Thanks!

furrtek, 2017-11-14:
Will have to unify the reception settings in a simple widget. Will do soon :)

unixpunk, 2017-11-14:
Ah ya, that would be good, same options on each Rx.  Thanks!

ImDroided, 2017-11-18:
I agree. I wonder if there is a easy way to do this. It would he awesome if we had a separate screen with a waterfall that would open for us to fine tune the received signal to get better results.

furrtek, 2017-12-08:
Added RF amp setting in 65a99bbe5ada43ad912986c51fd69fc3b4f17436

unixpunk, 2017-12-08:
Thank you, sir!  Will compile and play tonight!

unixpunk, 2017-12-09:
Works as intended.  In ADS-B Rx I can now toggle the Amp and see that it does take effect.  Thanks!

closed 2017-12-09 by unixpunk

djecom1, 2017-12-23:
@unixpunk I think your LNA is out of order. I had the same issue and I fixed it. You juste have to change the LNA with this method :
http://www.t4f.org/articles/repairing-the-hackrf/

unixpunk, 2017-12-23:
Thanks, ya I found that site as well and confirmed as much with his suggested testing.  I already have the new part just haven't put it on yet.  I believe GreatScott might also repair them too, though I have sought out a quote at this time.