furrtek/portapack-havoc #127: Add 1.245Mhz *LSE*

fantuz, 2017-11-23:
Would be nice to implement the PriceHa* software that I cannot find anymore anywere else :) I could never see or use it.

I am thinking of an hardware IR module to read price tags. Not to write anything new, just read (i.e. intercept updates or force retransmission...)

Just ordered a portapack :+1:

furrtek, 2017-11-23:
You want to use a Portapack as a 1.245MHz square wave generator ? I'm not sure that's a very efficient way to do it.

fantuz, 2017-11-23:
According to HackRF specs that should be doable, though an upconverter (that I own as well)...

True, inefficient and very expensive, plus risk to damage the board itself without proper filtering/coupling.

I would love to see a PriceHa* menu, would be something "new" to see some IR in the SDR panorama. I also look at https://github.com/cyborg5/IRLib2/ and few other options including Adafruit Si5351A. I also hooked up function generator but not so portable and I am not a PRO like you, @furrtek !

furrtek, 2017-11-23:
I'm definitely not a pro ;)
I'm not sure if it can generate a perfectly period-accurate square wave, those labels are quite picky about the length of each pulse. I tried using a 20MHz oscillator divided by 16 (1.25MHz) like they did in their transmitters. There are also very cheap AD9833 DDS boards on eBay.

Anyways, I'd like to be sure the RF output can be used to drive an IR emitter before doing anything.

fantuz, 2017-11-24:
or 9850/51 rught ?

furrtek, 2017-11-27:
Yes, I'm sure it'll be less trouble.

closed 2018-01-07 by furrtek