furrtek/portapack-havoc #128: Suggestion - Enhanced Direction Finding

darksidelemm, 2017-11-24:
The current PWM-based 'direction finding' mode isn't particularly useful. I've had more success working in USB mode and going based on volume.

Anyway, I've had good success with VK3YNG's 'sniffers': http://www.foxhunt.com.au/2m_sniffer/manual.htm
These represent signal strength using pitch, and automatically step in attenuators (with hysteresis) as the signal gets stronger, to avoid 'clipping' the pitch. An example of using one is here: https://www.youtube.com/watch?v=j0a8gWUJ0NA

Given the user already has good control over the gain stages in the hackrf from the 'audio' display, a pitch-based signal strength mode would be useful, and would turn the portapack into a fairly effective direction finding unit.
Optionally, a 'simple' step attenuator system (which I guess would just wind back the various gains in steps) would be a good enhancement, but the existing gain controls would do for now.

furrtek, 2017-11-29:
Thanks for the suggestion. The original PWM output was a quick hack for a friend, I agree it didn't have much use outside of what he was planning to do.
I changed it to pitch in d77337dd778d7638f83b3b761f1a22f1de283add, it goes from 1kHz to 3kHz depending on the RSSI value. It doesn't adjust the gain settings automatically for now. Also it's not well tested since I don't have directional antennas.

fantuz, 2017-12-12:
Hello @furrtek which frequency would you prefer be testing on ? I have plenty of antennas, I can easily borrow you a yagi/directional 800-960 Mhz 7 elements (and would be good for MIC testing too). I got 2 i can survive :+1:  :1st_place_medal:

fantuz, 2017-12-12:
in the HAM radio world, the direction finding is called "fox hunting" (chasse au renard).
Still fashionable nowadays, many people are interested in this game.