furrtek/portapack-havoc #130: FLEX Decoding

Pyr3x, 2017-12-04:
You seem to have support for POCSAG, any chance we can get FLEX support?

furrtek, 2017-12-04:
It can certainly be done but I won't be able to test with real transmissions. Maybe I can simulate some by feeding frames as audio to a PMR walkie-talkie ?

Pyr3x, 2017-12-05:
I can definitely test, FLEX is all over around here.

Pyr3x, 2017-12-05:
Also I could also record some transmissions and send them to you...

furrtek, 2017-12-05:
Maybe I didn't understand the specs correctly, but aren't the transmissions (batches ?) quite long ?

ImDroided, 2018-01-07:
Let us know what you need from us. I would love to see this implemented.

MischaBoender, 2018-11-11:
+1 on FLEX support. In the Netherlands FLEX is in use by the emergency services and know as the P2000 network.