furrtek/portapack-havoc #131: WideRange Jammer

iw3bti, 2017-12-05:
Hi, is possible to enable all the 3 range in Jammer function?

The suggestion is to hop between frequency ranges in 2 different modes:

**mode1**-cyclic-: start from range1, turn off range 1 and turn on range2, turn off range 2 and turn on range 3,turn off range 3 and restart.
**mode2**-timed-:start from range 1,  loop on range 1 for a time period setted in an specific option, turn off range 1, turn 0n range 2 and loop  for a time period setted in an specific option, turn off range 2, turn on range 3 and loop  for a time period setted in an specific option,turn off range 3 and restart from range1

thank you a lot!

chimera234, 2017-12-07:
On top of this, would be also possible to include the 5GHZ ranges as default templates? Same what is already available for 2,4 GHZ?

djjerdog, 2017-12-07:
+1 to both requests

furrtek, 2017-12-08:
I added support for ranges in frequency files in aba6c14074313e7deda6ba4b6d4f72ea4528b9c4. They're now stored on the SD card (JAMMER.TXT). The file also contains the 5GHz channels.

The cycling mode is next...

iw3bti, 2018-03-08:
any news about the cycling mode ?

zwt2135, 2019-06-05:
Any word on this?

furrtek, 2019-06-05:
Would have given some if there were any