furrtek/portapack-havoc #132: CTCSS tone code vs. frequency

iw3bti, 2017-12-05:
in the latest fw update (ctcss decoder) in the transmitters->microphone function, the ctcss frequency is missing. i think it's very usefull adding the freq. value near tone id and motorola id

furrtek, 2017-12-06:
Added back frequency display in 3221992ad1d5983b4aace5a076b44dcd0728617a

Pyr3x, 2017-12-06:
Did you correct the order on the options within the mic transmitter applet? It’s a pain navigating around at the moment as clicking down jumps around from top to bottom and top again etc..

furrtek, 2017-12-06:
You mean regarding the voice activation settings which get skipped when pressing down or up ?
The problem still remains, I just use the touchscreen for now but I will have to look at the widget focus code anyways.

closed 2018-06-12 by furrtek