furrtek/portapack-havoc #134: Multiple crashes going on

ImDroided, 2017-12-08:
Ok so I just compiled and the firmware is crashing on the following requiring me to hit the reset switch.

Transmitters:
Signal Generator
Soundboard
TEDI/LCR AFSK
And the touch tunes remote is now crashing after it loads the screen and I touch a button on the screen or if I try to use the buttons or dial on the portapack.

Great work.

furrtek, 2017-12-09:
Those modifications on the MenuView didn't go so well...

fantuz, 2017-12-09:
Here the behaviour i can see:
 - from the RX menu, you can always come back
 - sometimes you cannot go back to main after TX (and sometimes TX stuck ON)
 - sometimes after TX, even if TX might stop, no way to get back to main
menu.... this is quite constant as far as i tested
 - If you Record a file to SD, not going back to main menu even if you
correctly stop the recording.
 - replay, stuck as well, only after Replay *play button*

furrtek, 2017-12-09:
The problem is related to pointers to objects being deleted, so it's "normal" that everything goes crazy once that happens (at startup probably, since the menu is the first thing you see). I'm currently searching why that happens, sorry.

furrtek, 2017-12-10:
Is it better after 2d01822cdb0595d519910123a70a349e078e00fb ?
Was calling methods of inexistent objects on menu show/hide...

F0GPI, 2017-12-10:
hi furrtek, 
for your information,
in the last release, a bug appear on soundboard (freezing),
also on jammer and load frequency on all modes (tx&rx), and rolling to KDR444 (freezing) too.

fantuz, 2017-12-10:
i did get frozen on the SUBARU fob TX.
is not totally stuck but I can't go BACK in menu.

testing the rest of menus but seeing @F0GPI  comment, I guess some rework still needed.
Thanks @furrtek for all the support !

PS: bravo pour SSTV ++

furrtek, 2017-12-10:
@F0GPI Hello, the soundboard issue is a separate one which appeared earlier (#110).

I can load and save in the frequency manager but going to some categories (KDR444 or JAMMER) sometimes freezes on my unit also. I'm looking into that...

@fantuz I can't get the keyfob TX to freeze, even after transmitting. Is there a particular sequence you follow from the Portapack's startup ?

Thanks a lot for the reports.

furrtek, 2017-12-11:
There should be no more freezes in the frequency loading screen since 70c7646743ffc7af229bc8df8949b69feb23ecdc. Temporary fix...

fantuz, 2017-12-11:
Hello @furrtek I confirm it still happens in "key fob" after doing a "panic tx" short test, yes, still not able to go back, plus it really freezes.

I set the screensaver backlight to 10 minutes

fantuz, 2017-12-11:
burger pager does the same, latest firmware obviously.

PS: just for info, long time ago i have inserted a "blacklist hackrf" in my "modules.d" as I compile from sources. Might it be related ? Just wondering.

fantuz, 2017-12-11:
Hello @furrtek 

YES, basically every TX item still freezes for me (tried subaru, keyfob, lights).

Just for the note, another test I did was to recompile new hackrf sources (i was one or two commits behind, nothing major) and then re-uploaded latest firmware from hackrf built...

My environment haz latest HAVOC, latest HACKRF, latest GQRX and latest GR-osmo/etc just because i update more or less all the radio environment at once (including gnuradio).

Here the behaviour:
- GQRX works very well BEFORE re-flashing HAVOC.
 - then reflashed and "make program".
 - still, many TX are bad or freezing (but i can't say is worst than before as i took no notes) as we discussed few days ago
 - if i choose to go "hackrf-mode", the device will pop up correctly and be available to choice in GQRX, so i press PLAY and the acquisition will proceed as usual .... but .... all black and absolutely no signal coming in.... but is playing, so device correctly recognized etc.

I posted in #78 a guess about the hackrf hardware version, are we all talking about ANY portapack plus ANY hackrf, is it right? there exist different "hackrf versions" as one was the kickstarter/demo and one the final product, at least. Is portapack too, still the very same in terms of hardware ? I bought it few days ago from Rysc Corp.

Cheers !

F0GPI, 2017-12-13:
Thanks, the soundboard app works veryouell !!

fantuz, 2017-12-22:
See the original issue on portapack-hackrf to see what I mean -> https://github.com/sharebrained/portapack-hackrf/issues/115

Impossibility to go on hackr-mode correctly.
Device found, gqrx allows to start DSP, but is mute (you can change frequency and gain, but no way to "use" the hackrf outside of portapack-firmware.

Available to test

fantuz, 2017-12-24:
Persisting behaviour on Ubuntu 16 Ubuntu 17 (not tested on the Ubuntu 14 before release-upgrade).

Tested with boost 64 homemade, boost 63 and boost 62 from repos.
Also tested on boost 58, same issues.

HackRF mode doesn't work for me!

ImDroided, 2017-12-24:
I just compiled and everything is working for me. What happens if you flash
the original hackrf firmware?

ImDroided, 2017-12-24:
Or even the original portapack firmware

fantuz, 2017-12-24:
with original FW, works like a charm (no need to remove the portapack anyway)
flashing portapack (havoc or hackrf) is OK but HACKRF-mode is KO.

I also have clock IN (via GPSDO) so "hackrf_debug --si5351c -n 0 -r" returns 0x01 in place of 0x51

Portapacks HAVOC and HACKRF suffer the same issue to me.
Might the CLPD be different between different releases of one, jawbreaker and so on?

I own the final version of HACKRF (not the kickstarter) and 20170522 is the date on my portapack HW.

fantuz, 2017-12-24:
```
linux; GNU C++ version 7.2.0; Boost_106200; UHD_003.010.002.000-19-gd9561315
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-295-ga0adcd33
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Resampling audio 96000 -> 48000
BookmarksFile is /home/max/.config/gqrx/bookmarks.csv
gr-osmosdr v0.1.4-98-gc653754d (0.1.5git) gnuradio 3.7.12git-295-ga0adcd33
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Using HackRF One with firmware git-3447863 

```

furrtek, 2018-01-07:
I have the "old" 20140826 Portapack and I can confirm everything still works fine here with SDR# and GQRX.
Can you try going in the audio receiver in any mode, exit, and directly go in HackRF mode ?
Can you also do the same but just after having transmitted something ?
Is there any difference ? Still mute ?

jboone, 2018-01-15:
@fantuz I just noticed you are using an external clock input (from a GPS disciplined oscillator). Do you see the HackRF mode not work if you set up your system without the GPSDO attached? I am wondering if that is a significant variable that is different between your setup and ours.