furrtek/portapack-havoc #136: TETRA - add headers and SDS decoder

cnyxradio, 2017-12-09:
TETRA (CQPSK modulation) frames contains some headers :
- MCC (Mobile Country Code)
- MNC (Mobile Network Code)
- Colour Code
- Basestation Location Code

Also when a call is in progress, some useful informations could be decoded such has:
- the calling SSI
- the called SSI (individual or group)

Finally, TETRA supports data frames called SDS which can contains very interesting informations : GPS location, public transportations data, street lighting remote control...

lillstrumpa, 2020-01-03:
This would be really nice tog havet in the portapack.

lillstrumpa, 2020-01-03:
Nothing new !? Seems like alot of plugins and decoders for Windows and Linux have poped up since the the thread starten 2017 , I would love tog hand thos feature...

furrtek, 2020-01-03:
Maybe after making the effort to write in a decent manner you'll try to copy-paste those Windows and Linux decoders into the firmware and see how it goes.

I would love to have my breakfast made...

lillstrumpa, 2020-01-03:
May bad, Im sorry if it came out rude, not what I wanted... new tog havoc, Just saw the suggestion from 2017, just wanted to know if the suggestion was on ice...

lillstrumpa, 2020-01-03:
My bad, If I came out rude, it wasnt what I wanted...