furrtek/portapack-havoc #137: Jammer crashes when enable/load ranges

chimera234, 2017-12-10:
Hello,

Updated to latest version incl. added Jammer.txt. When enable/load ranges I get stucked in Freq. manager only reset helps to get out. BTW I also dont see the preconfigured 2,4/5 Ghz templates

Thanks
Chimera

chimera234, 2017-12-10:
When entering freq. manager I see category DMR/DPMR/FRS/GMRS. Once reached GMRS I cannot go back anymore, stucked -- Reset needed. As mentioned, beside above mentioned templates do not see 2,4/5 ghz templates.

Hope that helps

fantuz, 2017-12-10:
Hello chimera234, I guess you did not upload the content  of SDCARD folder ont the SD card base (root) folder.

Otherwhise the FREQMAN part (and maps, and other features) will be empty/unavailable as no config is present!

If you will not upload the proposed SDCARD contents, the files will be created on SD card at first usage, and empty content!

So -> get a SD memory card, will be useful to record/replay and take notes.

Cheers

chimera234, 2017-12-10:
Hello
Of course I have a sd card, all files are placed in the sdcard folder.
Anybody else see this problem or is it just me?

Thanks for let me know
Chimera

ImDroided, 2017-12-10:
Do you have the files in the root of the SD card?

chimera234, 2017-12-10:
Yes

chimera234, 2017-12-10:
When I enter freq. manager via "Utilities" no problem, I can see all frequency templates incl JAMMER.TXT, this indicates me that access to all files is given.

But when entering the Jammer I can just scroll to DMR/DPMR/FRS templates once I reached the GMRS template screen is frozen. very strange....... Any other idea?

Thanks
Chimera

furrtek, 2017-12-10:
I can reproduce the issue, I can't get passed the GMRS category (even sometimes in the frequency manager). As it sometimes works, I guess it's not a file parsing issue.

chimera234, 2017-12-10:
Thanks a lot for confirmation!

chimera234, 2017-12-10:
BTW, just made a donation because I think you are doing a great job here :)

furrtek, 2017-12-11:
Thanks @chimera234, I appreciate it a lot :)

I'm pretty sure it's a RAM issue now. The frequency manager doesn't crash because there's less stuff in RAM "behind". By going in the Jammer app for example, there's an additional menu screen and the jammer app itself still there, which (I believe) reduces the available RAM and prevents loading of the full list of frequencies from "large" files.

64kB isn't much so we're might be hitting a hw limit, but I'll have to check if nothing is bigger than intended or if anything is leaking.

For now the fix is just to cap the entries per file to 30, so I split the jammer ranges in multiple files in 70c7646743ffc7af229bc8df8949b69feb23ecdc

closed 2018-06-12 by furrtek