furrtek/portapack-havoc #139: Backlight setting is hardly retained.

fantuz, 2017-12-11:
The splash-screen preference is well retained
The backlight turnoff time is never retained...

The backlight duration setting is not retained within the same session, without reset ... this is weird, and inconstant behaviour. I set it at every boot, and screen goes in saving not in accordance with my preference (10 minutes vs 10 seconds).

I understand is a power-saving feature, but I would also say that a preference is a preference, and should take precedence!

Also, a "no screen saver" preference could be nice, if we really know the risk of doing it (this LCD will show memory effect very soon, but a preference should be set at own risk).

Cheers

furrtek, 2018-01-08:
The setting should be saved correctly now (587ef522f3e824903e97c06319a016240fdeff55).

unixpunk, 2018-05-21:
Confirmed saves and persists across resets and blanks (or not) on time as expected.

furrtek, 2018-05-21:
Thanks for trying it out :)

closed 2018-05-21 by furrtek