furrtek/portapack-havoc #141: CB Channels Added to FREQMAN

Pyr3x, 2017-12-12:
Can we add all the CB Frequencies to FREQMAN?

Example:

f=26965000,d=CH01
f=26975000,d=CH02
f=26985000,d=CH03
f=27005000,d=CH04
f=27015000,d=CH05
f=27025000,d=CH06
f=27035000,d=CH07
f=27055000,d=CH08
f=27065000,d=CH09
f=27075000,d=CH10
f=27085000,d=CH11
f=27105000,d=CH12
f=27115000,d=CH13
f=27125000,d=CH14
f=27135000,d=CH15
f=27155000,d=CH16
f=27165000,d=CH17
f=27175000,d=CH18
f=27185000,d=CH19
f=27205000,d=CH20

unixpunk, 2017-12-12:
I've got a couple files I've created, including this one.  What other ranges/uses would be of interest for US bands?

Pyr3x, 2017-12-12:
We should add the American POCSAG Frequencies too:

27.255 MHz | USA Unlicensed Part 95 (up to 100w PEP, shared with CB Radio Channel 23)
-- | --
35.200 MHz - 35.260 MHz (20 kHz steps) | USA
35.300 MHz - 35.540 MHz (40 kHz steps) | USA
35.560 MHz - 35.660 MHz (20 kHz steps) | USA
43.200 MHz - 43.260 MHz (20 kHz steps) | USA
43.300 MHz - 43.540 MHz (40 kHz steps) | USA
43.560 MHz - 43.660 MHz (20 kHz steps) | USA
72.020 - 72.980 MHz (20 kHz steps) | USA
75.460 - 75.980 MHz (20 kHz steps) | USA
152.0075 MHz | USA Medical/Emergency
152.030 MHz - 152.270 MHz (30 kHz steps) | USA
152.480 MHz - 152.840 MHz (30 kHz steps) | USA
154.625 MHz | USA On-Site Business Paging
154.640 MHz | USA On-Site Business Paging
157.450 MHz | USA Medical/Emergency
157.480 MHz - 158.670 MHz (30 kHz steps) | USA
158.700 MHz | USA
163.250 MHz | USA Medical/Emergency
454.025 MHz - 454.325 MHz (12.5 kHz steps) | USA
459.025 MHz - 459.325 MHz (12.5 kHz steps) | USA
462.750 MHz - 462.9250 MHz (25 kHz steps) | USA
467.750 MHz - 467.9250 MHz (12.5 kHz steps) | USA Low Power On-Site
465.000 MHz | USA
929.0125 MHz - 929.9875 MHz (25 kHz steps) | USA
931.0125 MHz - 931.9875 MHz (25 kHz steps) | USA

furrtek, 2017-12-13:
I'll add the listed frequency for sharing, thanks.
Know that you can simply add correctly formatted files to the /FREQMAN/ directory and it will be added to the categories automatically. The number of entries per file is limited to 30 right now because of the RAM issue but nothing prevents you from splitting the list in multiple files.

KC5CQW, 2017-12-14:
Is that why it has been crashing when I try to open it?

73, Damon

xmycroftx, 2017-12-15:
Welp....it must be that time-- time for circular buffers.

ImDroided, 2018-01-07:
I have added Chicago Police citywides and NOAA Weather. I am about to do CB 1-20 and whatever else I can find. Also added to the SD card a samples folder to use with replay

cornfeedhobo, 2018-08-14:
Could this issue get some attention?

cornfeedhobo, 2018-08-14:
I think this issue can be closed now.
See: https://github.com/furrtek/portapack-havoc/blob/master/sdcard/FREQMAN/CB.TXT

closed 2018-08-14 by furrtek