furrtek/portapack-havoc #143: Add austrian mobile phone frequencies into FREQMAN

chimera234, 2017-12-14:
Hello,

Because I am from Austria would be great to have austria mobil phone frequencies in FREQMAN.
For details please refer to below frequency assignment.

https://www.rtr.at/en/tk/FRQshare

Thanks
Chimera

chimera234, 2017-12-27:
Let´s close this - I added by my own - It´s anyway not something a developer should take care ... :)

Thanks
Chimera

closed 2017-12-27 by chimera234

fantuz, 2017-12-27:
hi @Chimera, what about creating a pull request ?

Closed #143 <https://github.com/furrtek/portapack-havoc/issues/143>.

fantuz, 2017-12-27:
Hello ... what about editing this file and push it to master (pending @furrtek approval) ?
-> https://github.com/furrtek/portapack-havoc/blob/master/sdcard/FREQMAN/POCSAG-2.TXT

For the benefit of the community !

Cheers @chimera234 and keep up the contributions!