furrtek/portapack-havoc #144: Implement circular buffers for display loading.

xmycroftx, 2017-12-15:
We are limited to a certain number of lines for the display area anyways, loading and displaying files can be managed by limiting the size of the buffer.  Since we can scroll forwards and backwards it should be circular (with the ability to pop or push from either end.). Since we don't have boost we'll need to implement the datatype.

furrtek, 2018-01-07:
A few commits back, the menu views (those which use the jog wheel) had a "menu item" object for each item. Those who were off screen were simply hidden.
I initially thought that this was causing the crashes, so I made those views to only instanciate enough "menu item" objects to fill the display, and update their text/icon according to the menu position.
Unfortunately I didn't notice a real improvement, I could list maybe 5 more items in menus before getting crashes again.

So in the end, I'm not sure if there's a leak somewhere or if everything is really eating that much RAM.
I'd like to figure that out before trying circular buffers. Have to learn some gdb...

furrtek, 2018-05-07:
I tried implementing circular buffers, they worked correctly for reading but things got very messy for editing/deleting :/