furrtek/portapack-havoc #145: Suggestion: Provide additional option to Antenna Bias Voltage

chimera234, 2017-12-15:
Hello,

I am using the HackRF with a LNA4ALL and always using the same Antenna. Everytime I switch of the HackRF I need to re-enable Antenna Bias Voltage. Would be great to have an additional option in there to get Antenna Bias Voltage enabled when switching on the HackRF

Thanks
Chimera

unixpunk, 2017-12-16:
To piggy-back on this, an icon in the top bar showing bias-t status would be a nice-to-have as well.  Not necessarily a toggle, just an indicator.  Thanks for your hard-work!

furrtek, 2018-01-07:
Hey,
I added a bias-T status icon in 3193c6ee99ca229cbfda374989c1fc6c4673385d. It can be made into a toggle if needed.
About saving the bias-T setting: is it safe to do so ? What if it's set to ON and the owner forgets to disable it before using hardware that doesn't support DC input ?

ImDroided, 2018-01-07:
Making it a toggle may be a bad thing. I can see myself forgetting it's on.
If you forced the firmware to disable it upon restart then have the icon
clickable with a pop-up with a confirmation screen might be a cool add-on

furrtek, 2018-01-08:
The icon is now clickable. Toggling it from OFF to ON asks for confirmation first.
The setting in the Radio settings menu is still present, the icon is like a shortcut.
(587ef522f3e824903e97c06319a016240fdeff55)

ImDroided, 2018-01-08:
Your the man. I sent you a email.

jboone, 2018-01-15:
@furrtek Good move. You're right, in that there's a risk that the antenna was changed between uses, and may be a DC short from antenna to ground. That's why I chose not to persist the bias between sessions. It sounds like your solution is a good trade-off between convenience and safety.

closed 2018-02-05 by chimera234