furrtek/portapack-havoc #146: tip for epar/xylos

F0GPI, 2017-12-19:
Hi, 
i have 3 cities around me on the same frequency who use bht for public lighting. 
Have you a tip for discover coding frames?

djecom1, 2017-12-20:
I use ScanScope as a CCIR decoder software for Xylos. For EPAR, I reverse the signal with a WAV viewer (audacity or..)

furrtek, 2017-12-22:
You're off to decode the FSK EPAR frames by hand, you should only need the city code (8 bits). See http://furrtek.free.fr/?a=epar.
For Xylos, you can also use Audacity with a spectrum view to match the tones to the symbols. The format is 0000CCFSRRBSSSSB0000 with CC=City code, F=Family, S=Subfamily, RR=Receiver ID, S=Relay states (0=Ignore, 1=Open, 2=Closed). Same consecutive symbols are coded with E (0000 becomes 0E0E) and A can be used as wildcard.

chownsauce, 2018-05-10:
Anyone know if Epar/Xylos (CCIR tones) are mainly only used in Europe? I'm trying to figure out what's used in the USA but I think it's just controlled by a sensor that detects daylight. :/

furrtek, 2018-05-11:
I'm not aware of any city outside of France using Epar or Xylos. I know many cities use either light sensors or digital astronomical clocks (sometimes synced by GPS).

StoneSet, 2019-07-01:
Hi,
There are only 255 possibilities (so 8 bits), so you can try modifying the code of @furrtek project from 2012 (http://furrtek.free.fr/?a=epar). As soon as a command is sent, there are 255 possibilities tested and once the cities code is known you can directly inform the emitter without "brut force" the receiver, then, it will go faster to shutdown the area. 
I will try this technique myself, I come back to you as soon as I have results.
Thanks