furrtek/portapack-havoc #147: Question: Scanner availability

chimera234, 2017-12-27:
Hello,

Based on current schedule, by when is it planned to have a scanner feature available?
This feature would be really great to have on the portapack as well, maybe I will not use my 
Icom R-20 that often anymore.....  :)

Thanks for an update on this
Chimera

furrtek, 2018-01-07:
Hey,
By "scanner", you mean an app that would continuously check the RSSI of multiple channels and stop when one exceeds a threshold ?

ImDroided, 2018-01-07:
Would be cool if we could scan the freqs in a file from the sd card

unixpunk, 2018-01-08:
+1, exactly as described, like a police scanner, cycle through a freq file on sd card, pause on high RSSI w/audio ON then wait N seconds after RSSI goes back down before going back in to 'scan' mode.  N=2 seconds would by my suggestion as that is what my police scanners all default to which I've not needed to ever change.

Could also have the current search as a second mode, where it scans a band from Xmhz to Ymhz with Skhz steps between frequency stops.  Same stop w/audio on for high RSSI, N sec pause after RSSI back down before searching again.

ImDroided, 2018-01-08:
Based on his radio he mentioned unixpunk hit the nail on the head.

ImDroided, 2018-01-08:
Based on the radio chimera mentioned it sounds like unixpunk hit what he was looking for.

chimera234, 2018-01-13:
yes exactly,thanks

ImDroided, 2018-01-13:
This would be a awesome to have. I spend a lot of time out of state and this would be cool to scan in the area. Out of all suggested features this one would top of my list.

ImDroided, 2018-02-04:
Any update on this feature? I know this is a big one to code but it sure would be sweet.

furrtek, 2018-04-19:
Added a test app in 3ddc6553ace69bd0c2219e78ebab4c7b38c21ef8.
The frequency list can't be changed for now. If you want to try it out, you can edit /firmware/application/apps/ui_scanner.cpp around line 111. Remove the `frequency_list.push_back`s and put yours in. Frequency is in Hz and you should be able to add a bunch without crashing the app.

I'll add list loading later on. Right now I'm having issues with audio not unmuting.

furrtek, 2018-07-25:
Hello,
As with the other parts, I'm still doing this on my free time. To get something fully released, I need to fully finish it first (which it isn't). Since my main goal these days is to stop living on $100, I don't have that much time to dedicate to the firmware, and I can't plan on anything regarding that.

So in short: I know I haven't worked a lot on it since a while, I still want to, but I really have to focus on things that can bring me food on a regular basis. I'm very grateful for the donations I received and I thank the people who donated again, it really helped. But I can't count on that to live...

chimera234, 2018-07-28:
Hello furrtek,

Thanks for your comments to explain the reasons why not much is going on since a couple of months.

You are right in regards to donations, sometimes little bit is coming in, but nothing to rely, no certainty what will come in next month...

What about to change that? If ok for you I would like to ask all registered users who is willing to pay every month, let's say 10 euro. 

I guess the majority can afford to pay 10 eur/month - Would be great to see how many others users are willing to join.

Is this something that will help you and agreed from your side?

Thanks for let me know
Chimera

RED29, 2018-07-31:
@furrtek see my comment on issue #193 
Set up a Patreon account and a lot of people will donate each month. I will be the first. Your tech is awesome.

HWpirate, 2018-08-01:
furrtek, Please do open an account, I would be one of many in line to contribute.

ImDroided, 2018-08-01:
I've donated before and I'll do it again. If I had more time in life I'd
dive into this myself but with work and family my computer time is severely
limited these days

mwarchut, 2018-08-02:
Yes.. Please setup a patreon.. Ill sign up for a monthly donation for sure..

furrtek, 2018-08-02:
To all: thanks for your kindness.

Without going to far into personnal stuff, I should explain that my problem isn't just about money but **how** I can earn money on the long term without going crazy.

After a few experiences, I believe that I'm quite different from most people regarding that. In the way that I currently can't accept money for something that isn't done and working yet, or something I'm not 100% sure of being able to do.

Basically I'm driven by interest and motivation, but I don't really control those.
Even if I could get $20, $200 or $2000 monthly from Patreon, it wouldn't ensure I'd have enough courage to work on the firmware. That's how much I'm affected.
I can accept money as a reward, but not as a way of pushing me to do things, as often it does the opposite (because of the stress, fear of not finding solutions, fear of beeing paid too much, ...).

That's why, for now, I'd rather not do this.
I'm currently looking for a "therapeutic" job which pays me to do something that always work, to see if that suits me better than creative stuff (at least to make a living).

Unfortunately this means there's no reliable way of pushing me to work more on the firmware, but maybe things will change with time.

Bobbersonxd, 2020-04-11:
I hope your feeling up to updating more of the FW. Im a huge new fan. :) Great Work

furrtek, 2020-04-11:
Thank you but honestly... no.
With the flood of messages from clueless people still coming in, motivation actually got worse.