furrtek/portapack-havoc #148: NOT ISSUE:  Would anyone else like to see a HAVOC facebook group to discuss success/fails on their attempts?

chudgoo, 2018-01-01:
Seems like the GitHub issue board is becoming the trial and error zone. 
Would anyone else like to see another forum (that doesn't clog up Furrtek's inbox unnecessarily) while we all work out our smaller problems?
I'll start one and hope to see some of you there with your problems, solutions, work-arounds, etc.
The tool is amazing. The capabilities are amazing. The 'getting shit to do what you intend' is a crap shoot sometimes and help is always out there... Just not sure the GitHub "Issues" page is the place for a lot of it.
Delete if you think I'm out of line btw...

furrtek, 2018-01-01:
Maybe link to it so I can share it ? I can't find it from Facebook's search engine.
I think it's a nice idea, and if an issue or an idea needs tracking, I can always open an issue here to add technical details about progress.

unixpunk, 2018-01-01:
I'd join and help out where I can, as long as its not on facebook, I'd suggest something else like groups.io, etc though.

ImDroided, 2018-01-01:
I'd join. That's githubs one flaw... No discussion area.

azzeloof, 2018-01-01:
How about Slack?
(Also, Hi. I haven't contributed at all yet but I'd like to!)

chudgoo, 2018-01-01:
https://www.facebook.com/groups/177623356165819/

fantuz, 2018-01-03:
hi, such facebook page is not available for me. what about permissions? my fb address is different from my github one.

i see that groups.io was alsochoosen by sdrangel project for thesame purpose: shall we also move there as proposed by @unixpunk ?

ImDroided, 2018-01-03:
Furrtek you should make the page wherever it may reside so you can remain
in control I'd hate to see us get locked out due to a random person
abandoning the page.

chudgoo, 2018-01-03:
I only suggested facebook because it's becoming very ubiquitous.
As for abandoning the page, aside from creating it, I had no intention of
playing over zealous moderator or anything... or shutting it down for any
reason,
I just want to talk shop in a members-only, private forum with other
mischievous types and keep the shop talk off the github page. (which I
agree REALLY does need a forum feature)

xerospawn, 2018-01-04:
The facebook link says page is not viewable at this time.

chudgoo, 2018-01-04:
Ugh. My mistake... it was set to secret group. It’s public now.

ImDroided, 2018-01-06:
Chudgoo..... I only mentioned that because of a situation I am currently in. I use to use a piece of software and I created a group discussion for it. Fast forward a few years and I dont use said piece of software anymore but the group is still active and I have no was of passing the group to someone else to take care of. I still have to log in every so often to do things that I now dont want to do.

furrtek, 2018-01-07:
Joined the FB group (Frna... that's my ROT13'd name).

h4waii, 2018-01-10:
Why not Groups.io as previously mentioned, or Slack, Discord, Telegram, even Reddit? They're all much more "open" than Facebook yet still allow users to remain fairly anonymous.

fantuz, 2018-01-27:
hello @h4waii come join the group on FB, anyway nothing much is happening (we're less than 20 so far)

closed 2020-05-24 by furrtek