furrtek/portapack-havoc #151: Replay wont load file after file rename

ImDroided, 2018-01-08:
Ok so after renaming the capture file I recorded of my fans on off remote now replay wont load the file. I also renames the txt file the same. Is the filename hard coded in the file itself? What would cause it to not load after a rename?

ImDroided, 2018-01-08:
If I copy the orginal file to the samples folder it loads just fine but rename the file and txt file and it fails

furrtek, 2018-01-08:
I tried recording to "BDD_0003.C16" and replayed it back OK.
Renamed it to "BDD_ABC.C16" and replayed it back OK.
What name did you give to the file ? Did you keep the .C16 extension ?

ImDroided, 2018-01-08:
It is the file to control the fan in the sample dir. I can load the file just fine before a rename. Yes I kept the C16 file ext. go copy that file from the samples dir and try to load it

ImDroided, 2018-01-08:
Ok so I just copied 2 sets of the org file to the samples dir on sd card.  I then renamed one set from BBD_0009.C16 to BDD_Fan1.C16 and did the txt file the same. replay will not load the renamed file but loads the orginal just fine.

ImDroided, 2018-01-08:
Here is the original file 

[BBD_0009.TXT](https://github.com/furrtek/portapack-havoc/files/1610520/BBD_0009.TXT)

And here is the renamed one.

[BBD_fan1.TXT](https://github.com/furrtek/portapack-havoc/files/1610526/BBD_fan1.TXT)

furrtek, 2018-01-08:
It should be fixed in 8f9c75d62e01b714d9fd570ac2a5e3204a655abe.
The file loader screen wasn't returning the full path... Thanks for the details !

ImDroided, 2018-01-08:
Ok it loads the file now thanks. Any chance of having the replay look for the text file to grab the freq needed and whatnot?

furrtek, 2018-01-08:
Added in aebd1757da33f07c55868bce9c4fbb2dc39e9978. It should load the record frequency if there's a .TXT file with matching name in the same directory.

ImDroided, 2018-01-19:
confirmed working.  I like that it loads up with the file.

closed 2018-01-19 by ImDroided