furrtek/portapack-havoc #153: Freezing when accessing Load Range or Load with NOAAWEATHER.TXT

djjerdog, 2018-01-08:
After updating my SD card with the latest FREQMAN files Portapack is freezing whenever I try using "Load Range" or "Load" to select a frequency. 

I narrowed it down to NOAAWEATHER.TXT, after removing this from my SDcard no more freezing issue.

furrtek, 2018-01-08:
I tried it, copied the NOAAWEATHER.TXT file to the FREQMAN directory and couldn't get it to freeze. I can load the frequencies from the audio receiver and the frequency manager screens.
How many .TXT files do you have in FREQMAN ?

ImDroided, 2018-01-08:
Have you compiled the latest build?

djjerdog, 2018-01-08:
Yes, using the latest commit, I tried formatting the SD card and reloading all the files a couple of times and had the same freezing issue. After investigating further I don't believe it was specifically to do with the NOAAWEATHER.TXT file, because even after I removed that file and added more from the list I still ran into the freezing issue. I've not removed all except 4 of them that I primarily needed and haven't experienced any more freezing issues. Perhaps it has something to do with the number of files?

furrtek, 2018-01-09:
I have 18 files in the FREQMAN directory, tried different operations (save, load, rename) and couldn't get it to freeze.

chimera234, 2018-01-13:
Hello,

Same issue here. It happens as follows:

First time when loading s frequency from NOAAWEATHER it loadind successful, no issue.
2nd time when loading another frequency from NOAAWEATHER, nothing happens anymore screen is freezed.

Thanks
Chimera

chimera234, 2018-01-13:
Just rechecked, freeze happens every time as soon I reload another frequency from FREQMQN.
First time frequency loading always successfull.

Thanks
Chimera

chimera234, 2018-01-14:
Hello,

Anybody else has same issue as described above?
Is the solution just to reduce files in Freqman?

Thanka
Chimera

unixpunk, 2018-01-14:
I have this with fewer files too.  For me it let me load 2 freq's from file before it froze on the third attempt as soon as i clicked load that third time.  I will grab and compile from latest source and test once more.

unixpunk, 2018-01-14:
Using latest source I was only able to load 1 freq, then on the second click of Load, it froze.  Doesn't matter what Freq, I just picked the first one in the first file, BHT-something.  Last time I used CB 19 and then CB 9 before it froze the third attempt. HTH

chimera234, 2018-01-15:
Thanks for confirmation, let's wait for additional comments from furttek how to proceed.

chimera234, 2018-02-06:
In regards to this issue I mitigated it by deleting some of the list which I not need, afterwards no issue anymore.

blakekrone, 2018-03-01:
I too have been experiencing this issue. If I remove majority of the files and leave just a handful, for me the WLAN frequencies, it works. I've tried with both a small (2GB) and large (32GB) card. If I leave all the files present from the repo the minute I hit load in Jammer it freezes. I can view the files in file manager and frequency manager without any problems though.