furrtek/portapack-havoc #154: Any Guidance on Proper SD Card Loading of Files

wirednot, 2018-01-08:
Sounds like a silly question, I know- please tolerate my newness to all this. 
-Have created the folders on SD Card, like FREQMAN. 
-Have copied a given frequency set, like FRS for example, into the FREQMAN folder. 
-When I summon FRS (or any other group) in FREQMAN, it's easy to find- but all individual frequencies you'd expect to show are just 0.000M. 
- Not sure what I'm missing to actually see the discreet frequency values?

Am running the latest pre-compiled available version.

Thank you.

furrtek, 2018-01-09:
Are you using files directly copied from the repository ? Or did you edit them or save them back ?

wirednot, 2018-01-09:
straight copy- no edit.  Remove SD from Portapack- put in Mac. Create each folder (FREQMAN, etc) then within FREQMAN, copy the individual files straight out of repository.

obinrobinson, 2018-03-19:
I have the same issue as well. I feel like there's something obvious I'm not seeing but any help would be greatly appreciated.

chudgoo, 2018-03-20:
I too am having issues with freqman and sstv, but was too afraid to ask.

ImDroided, 2018-03-20:
It's a known issue.

furrtek, 2018-03-27:
Are you letting the splash screen display on startup ?
If yes, try disabling it in Setup -> UI and see if the problems are still frequent.
I'm not sure but I think there's a lot of UI stuff staying in RAM.

KC5CQW, 2018-04-27:
Do not keep the “sdcard” folder. Move all contents from “sdcard” to the root of the actual sd card.

furrtek, 2018-05-15:
I never got the 0.000M frequency display problem, but 96828262d712b08abb675b7c4424fec47d8ffb93 might have solved it.

unixpunk, 2018-05-21:
I've never experienced this issue and an able to see/load freq's no problem.  If anyone has better steps to recreate, happy to test.

closed 2018-06-12 by furrtek