furrtek/portapack-havoc #157: Capture bandwith always fixed at +/-200k

chimera234, 2018-01-13:
Hello,
As far as I remember in previous versions the channel bandwith in the capture feature got set according to the setting in the audio receiver, but looks like this is not the case anymore.

How can I change the channel bandwith in the capture feature?I am currently unable to change it, it is fixed at +/-200k

Thanks
Chimera

furrtek, 2018-01-13:
Hey,

The audio bandwidth does change depending on the receiver mode you're in, but I believe the I/Q capture bandwidth has always been +/-250k.
I read Sharebrained original code again: sampling_rate is set at 4MHz and the data is decimated by 4 then by 2, so 4M/4/2 = 500k.

It would be possible to add a setting for a few steps down (500k, 250k, 100k, 50k, 25k, 12.5k ?). Should I add that ?

ImDroided, 2018-01-13:
That could be useful.

chimera234, 2018-01-14:
That would be really a great feature.

Thanks so much!!

furrtek, 2018-02-22:
In 7fd987a2b4ca2b746b2ed774810e1725ac8c1f38, steps are 500k, 250k, 100k, 50k and 25k.
The value is written in the .TXT file, and is loaded automatically during Replay like the frequency.

chimera234, 2018-02-22:
works,thanks

closed 2018-02-22 by chimera234