furrtek/portapack-havoc #158: Microphone TX - Mic is not working

chimera234, 2018-01-15:
Hello,

Just wanted to use microphone TX feature and figured out that Mic is not working. Tested with 3 different Mic´s none of them are working - green volume bar is not moving. Tried to use it with different mic gains, no luck still not working.

Does the microphone TX feature work for anybody else?
Thanks for let me know

Chimera

jboone, 2018-01-15:
Hi @chimera234. Are you using a smartphone-type headset? The PortaPack connector is compatible with most smartphone headsets, and uses a four-conductor TRRS connector. You can read [more on the PortaPack wiki](https://github.com/sharebrained/portapack-hackrf/wiki/Hardware).

chimera234, 2018-01-15:
Yes, I tried with smartphone headset, also checked with a single mic but no success. 
As far as I remember it worked some time ago before I updated, thats why I wanted to know if someone else has same issue.

furrtek, 2018-01-16:
Have you seen #123 ?

chimera234, 2018-01-16:
Thanks. In my case even after exiting the app mic does not work.

jboone, 2018-01-17:
@chimera234 Which version or git commit hash of firmware are you using? And what is the version number on the front face of your PortaPack?

chimera234, 2018-01-18:
Works now as described - I misunderstood "exiting the app" - I always reset the HF and tried again, of course without success. Now when starting the mic TX app, I simply exit via the "go back" arrow and enter again and voila mic TX app works :)  Thanks a lot

furrtek, 2018-02-01:
Should be fixed in 57c759627d60b3716f0dafc150d622423b9eafe5.

chimera234, 2018-02-05:
works for me :) Thanks a lot

closed 2018-02-05 by chimera234