furrtek/portapack-havoc #159: No Bias voltage present at output (LNA4all TX modification)

chimera234, 2018-02-05:
Hello,

Just wanted to use my modified LNA4all (~100mW Output) but I figured out there no amplification at all, same level as without Bias Voltage. Then I measured the Output voltage directly at the HackRF output (without LNA4all installed) and figured out that I do not have any output voltage.
Would be great if somebody can confirm if output voltage is present, should be 3,3 Volt.

Thanks a lot
Chimera

furrtek, 2018-02-05:
Hello,
I can confirm there's no DC voltage in either case, when using the top icon or the settings menu. I'll look into it quickly. Thanks for the report !

djecom1, 2018-02-06:
The previous build works fine : https://www.youtube.com/watch?v=lbADVqBy9FE

furrtek, 2018-02-22:
It should be fixed in 7fd987a2b4ca2b746b2ed774810e1725ac8c1f38
Tested with voltmeter, when entering/exiting views and toggling during rx/tx.