furrtek/portapack-havoc #160: REQ: Long Range Systems - Restuarant pagers?

chudgoo, 2018-02-10:
I went to Panera Bread the other day and they handed me one of those pagers used to tell you when your order is ready. I immediately looked up the FCC info and found it uses 467.75MHz.  
https://fccid.io/2AB6ORXCS7

Also it seems someone has done some leg work decoding it and it looks fairly straight forward.  POCSAG at 2400 baud and uses FSK.
http://www.windytan.com/2013/09/the-burger-pager.html?m=1

It would be fantastic fun (once or twice anyway) to light up the whole stack.

(That said, I’ve never worked in retail/food service, but can imagine I’d be super confused if this happened on my shift...  one of the commenters on the last link had a better idea. When waiting for a table, set yours off early and see what happens. Before anyone gets mad, the firmware is called “havoc”, is it not?  ;)

ImDroided, 2018-02-10:
Hahaha yes red Robin has these also I would have fun with this

furrtek, 2018-02-22:
POCSAG RX and TX are already done, so all that would be needed is a special POCSAG TX mode which would transmit address-only messages in a defined range ? Or maybe -50/+50 around a given address ?
A friend noticed that some MMCall pagers used POCSAG also, but the messages had a text payload.

jLynx, 2019-07-26:
@furrtek this video explains how LRS works https://www.youtube.com/watch?v=ycLLb4eVZpI
He does it on a HackRF
I would love to see this implemented in Havoc (as in being able to specify the pager ID and restaurant ID)