furrtek/portapack-havoc #161: What is included in upcoming SW release?

chimera234, 2018-02-11:
Hello,
Is it already fixed what upcoming sw release will include?
My wishlist :)

- Bugfix Bias voltage
- Flexible settings for Capture bandwith
- Scanner feature

Thanks for let us know :)

Chimera

furrtek, 2018-02-12:
Hello :)
The bias voltage bugfix and more settings for capture are on the top of the list. The scanner and finally getting APRS RX to work is next. I didn't work much on the Portapack since the beginning of January because some other project got a huge boost and I can't stop working on it :< (for now).

ImDroided, 2018-02-12:
What's this other project? Anything we might be interested in? You cheating on us bro? Lol if it's radio related post a like. Unixpunk and myself have been playing with our plutosdr's lately.

furrtek, 2018-02-12:
Totally unrelated to radio, I'm not cheating you ;p
It's about reverse-engineering custom chips on a gaming system, infos: https://www.patreon.com/furrtek
I still want to add plenty of stuff to the Portapack, won't giving up on it soon !

ImDroided, 2018-02-12:
Lol just giving you some crap. That is awesome. I am currently sourcing parts to build a raspberry pi based old school arcade console. So far I have found the cabinet, vinyl  graphics for the cabinet ,RPi, buttons and joystick, only thing left is to decide what monitor to use and what button configuration I want them to ship the cabinet with. Because it's RPi based I am also ordering USB retro controllers for Nintendo, SNES, Sega, and whatever other systems I can find. Should be cool.

chimera234, 2018-02-12:
Ok, thanks for the update, just hoping it will not take months to get you back here :)

furrtek, 2018-02-22:
Bias voltage fixed (I hope) and added 5 capture rates in 7fd987a2b4ca2b746b2ed774810e1725ac8c1f38 :)

chimera234, 2018-02-22:
Thanks, both features are working without any issues :+1:

chimera234, 2018-03-16:
Hello Furrtek

Can you already let us know what will be included in next update?

Maybe already to have Scanner available? 😊

Thanks
Chimera

closed 2018-05-15 by furrtek