furrtek/portapack-havoc #163: Please share your recorded samples

ImDroided, 2018-02-18:
Hey everyone I am working on putting together a collection of playback files for known devices. If you have any recordings of any devices please let me know.. I would like to get as many devices as possible in the samples folder for fun reasons while out and about. I have a few that I will be adding but would like whatever you may have.

maulwurf98, 2018-02-18:
This is a great idea. As written in my feature request before, one could boost this playback option with extracting from pilight.

djecom1, 2018-02-19:
I found this : [](http://www.kb9ukd.com/digital/)

furrtek, 2018-02-21:
Just for info: Github doesn't like very large files. There's a 100MB limit per file and a whole repo should be below 1GB.
100MB is roughly equal to 52 seconds of I/Q data recorded at 500ksps, so trimmed files should be fine. Please just avoid leaving 10 seconds of noise before and after a 0.1s frame, for example.

ImDroided, 2018-02-21:
No worries I have a website I'm setting up http://www.rfoverride.com I plan
on hosting all this stuff there in a nice list

jLynx, 2019-07-26:
@ImDroided any progress with your website? It seems to be down for me

ImDroided, 2019-07-26:
I forget to renew my hosting plan I still have the domain and I'm pretty
sure I can recover the hosting. I'll look into it and get back to you.

pawel37, 2019-12-23:
https://github.com/richalonso/hackrf_havoc_map

pawel37, 2019-12-23:
> 
> I forget to renew my hosting plan I still have the domain and I'm pretty sure I can recover the hosting. I'll look into it and get back to you.
> […](#)

https://github.com/richalonso/hackrf_havoc_map