furrtek/portapack-havoc #164: Video camera sniffing

silascutler, 2018-02-21:
Would be awesome to add support for receiving 900MHz - 2.4GHz RF cameras.  These often encode the video streams with either NTSC or PAL. 

Related projects / Uses:
https://makezine.com/2008/02/27/sniff-wireless-video-came/
http://www.recyclism.com/twopointfour.php 

Related GitHub Projects:
https://github.com/argilo/sdr-examples/tree/master/ntsc
https://github.com/gnuradio/gr-etcetera - gr-ntsc.lwr
https://github.com/kik/sdr-tv
https://github.com/seminar/televison_back

silascutler, 2018-02-21:
Also this project too: 
https://github.com/lscardoso/gr-ntsc-rc

furrtek, 2018-02-22:
I'm pretty sure receiving realtime video won't be possible on the Portapack since drawing to the screen is quite slow.
I'm not even sure it would be possible to record frames directly to the SD card without showing them. The 7MHz channel could be "seen" alright, but processing might be too demanding for the mcu.

If there's enough CPU power then it could be possible to show a frame once in a while, and if not an NTSC/PAL analog signal detector can at least be done.

(Cool links !)

chimera234, 2018-02-22:
+1 Great Idea :)

tibas21, 2018-04-02:
Furtek you're right. I have an hardware camera detector, the lawmate compact camera detector/monitor. I reverse it to know if we could add this techno inside the portapack ans this is complicated. The portapack is not enough powerfull, and the screen not optimised for that.
But.. Maybe an other system could be created. Don't know but i'm interested about that.

chudgoo, 2018-04-02:
Would the I/Q data piped out to an external receiver or monitor work instead?

geeknik, 2018-07-20:
Maybe instead of streaming the sniffed video we just capture a frame from it and save it to disk as a PNG or something?

jamesshao8, 2019-11-18:
We have achieved analog tv demodulating using hackrf on a host computer. I think to demodulate black and white signal, 2MHz is more than enough, since TVSharp is using rtlsdr to do the job. Also the grey scaled signal is AM signal, we will try on a computer to see the lowest cpu usage we can achieve and then think about bringing to portapack. Maybe we can drop some resolution to barely see what's happening.

jboone, 2019-11-20:
@jamesshao8: For what it's worth, most FPV and security cameras I've encountered use FM to modulate the video signal, not AM as used by NTSC broadcast TV. So the band-limiting trick not might work as well.

jamesshao8, 2019-11-20:
@jboone Thanks for your comment. In my country, the analog fpv camera for drone are usually PAL. I firstly demoluated one of them in TVSharp. You can see in this video below (0:28~1:00), the left side window is tvsharp (grey scale), the right side is showing the image from its original receiver.
https://v.youku.com/v_show/id_XMTc4MDgzNTkwOA==.html

In the video, I am using a FPV camera in 1.2GHz which is inside the frequency range of rtl-sdr. However most of the camera I can find is 5.8GHz, so I made a similar program with HackRF under linux directly calling its API. I decompiled the source code of TVsharp, it's calculating the magnitude of IQ sample, hence AM demodulating. So I did the same with HackRF. And it's working. The only problem with my program is it's always rolling. However it's enough to tell what's in the image.

jamesshao8, 2020-02-25:
Hi @silascutler @jboone  I am implementing analog tv receiver in portapack, and I have some achievement now.

We are discussing here:
https://github.com/jamesshao8/portapack-havoc-modified/issues/2