furrtek/portapack-havoc #168: Add additional bandwith settings to "Capture" feature

chimera234, 2018-04-01:
Hello

Please add as well additional bandwith settings to "Capture" feature, same that are available in "Audio"
8k5
11k
16k
Currently the lowest available is 25k, having issues to properly tune into signals when using the "Capture" feature.

Many thanks in advance
Chimera

furrtek, 2018-04-18:
Options added in 563676422603400e391dbbd7df4fc84168d4c8f6

chimera234, 2018-04-20:
Thanks a lot! One question: Why is the speed of the waterfall view that high at 11k compared to the other settings?

Thanks
Chimera

furrtek, 2018-04-20:
That might be a serious bug. I'm not seeing the same kind of patterns on the same frequency at 11k, and other sample rate. I never touched the waterfall code so I probably missed something.

chimera234, 2018-04-22:
I just checked again on my side, when switching to 11k the waterfall speed is probably trippled.
Anybody else can confirn?

Thanks

ImDroided, 2018-04-22:
I will test it when I get home later.

furrtek, 2018-04-22:
It does goes faster on my side too, I mean. I just don't know why that happens yet :)

jboone, 2018-04-22:
My recollection is that the spectrum row output rate (scroll rate) is directly proportional to the sample rate of the output of baseband processing. I’ll review the code in a few hours (when I’m next near a computer).

        - Jared

chimera234, 2018-05-02:
Hello Jared,
Did you already check the code?

Thanks
Chimera

unixpunk, 2018-05-21:
Late reply here, but I see the same behavior.  Everything shows the same scroll speed except 16k, 11k and 8k5.  (respectively, faster, fastest, fast than all other higher samples)

furrtek, 2018-05-21:
Taking care of this right now. The scrolling speed shouldn't impact the recording, maybe the current consumption at worst.