furrtek/portapack-havoc #175: Bug? Screen blacks out after starting a few things

ImDroided, 2018-05-16:
Ok so under the Keyfob, adsb transmit, jammer... pretty much anything with a start button the screen blacks out upon hitting start. Is this a bug?

furrtek, 2018-05-16:
Can't reproduce, did you build from source recently ?

unixpunk, 2018-05-17:
I don't have this issue.  Mine is Git hash 2d3a631.  arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release)

I'll pull/flash the latest now and test that.

ImDroided, 2018-05-17:
> Still works fine, hash is now b11c3c9

Ill rebuild and test again.

ImDroided, 2018-05-19:
rebuilt and still doing it. built from my desktop this time. upon clicking the srart button the screen goes black until I hit a button then it comes back.

furrtek, 2018-05-19:
Don't you have stealth mode enabled (mask icon on the top bar) ?

ImDroided, 2018-05-19:
Well don't I feel like a dumb ass. I don't know how that got enabled.

closed 2018-05-19 by ImDroided