furrtek/portapack-havoc #180: Add PL subtones

F0GPI, 2018-06-01:
Hi, 
It is possible to add the pl subtones on audio receptor (decoding)and transmitter modes?
Like the ctcss but the pl subtones is mostly used on professional pmr. 
Thanks.

kevtheskin, 2018-06-02:
Hi F0GPI
I think PL tones and ctcss are the same . Motorola call them PL tones . see https://en.wikipedia.org/wiki/Private_line .

Cheers 73s MM0KJG

furrtek, 2018-06-03:
Hello,
They're already there. See the two characters at the end of the tone designator.

F0GPI, 2018-06-06:
Ok ctcss and PL is the same. 
I’m confused for this noob error.
But another coding tone squelch named tone on most pmr /ham radios, it’s this coding Squelch who said. It si possible to add this mod for decoding and encoding?

furrtek, 2018-06-06:
Are you talking about DCS ?

F0GPI, 2018-06-08:
No not dcs. 
Thanks for the reply. 
The problem is solved. 
I understand the two caracters are the code of the tone and the two Numbers are the ctcss code. 
I close this issue.

closed 2018-06-08 by F0GPI