furrtek/portapack-havoc #190: Pocsag update

kevtheskin, 2018-06-17:
FAO Furrtek,
Hello there, 
I think I know why the decoding of the pocsag is so intermittent. I ran the unit in hackrf mode on pc and tried decoding with Pdw and Sdr#.   I noticed that when you apply any gain which I do require at my locality I see lots of mirroring of the signal either side of the spot frequency which seems to stop the decode . I compared it to my Rtl sdr and sdrplay which was ok at 100% decode. I also tried my spare hackrf and it displayed the same mirroring. Would it be possible to add a filter to help reject the residual noise.
Cheers Kev

kevtheskin, 2018-06-17:
Hi Furrtek,
Can you tell me why you marked this as duplicate please?

Cheers Kev

furrtek, 2018-06-17:
Hello. Because it's related to #178. It's just to keep track of the same issues.

closed 2018-06-17 by kevtheskin