furrtek/portapack-havoc #192: soundboard transmitter not detecting wav files

richblackhat, 2018-07-20:
i have in the sd card a directory called wav  
i have inside lots of wav files, when i enter soundboard it shows me no files in wav directory

it would be nice if the audio reorder save wav files directly in the wav folder to use them later , i had to copy the files manualy on my computer

am i doing something wrong or it is a bug  
in the root of the sdcard i have the wav folder with files on it. i wonder what happens

furrtek, 2018-07-20:
Are the files in the correct format ? https://github.com/furrtek/portapack-havoc/wiki/Soundboard

I think recorded wav files can't be played back because they're 16 bit (which is stupid, I agree).

MrARM, 2018-08-07:
Yeah, its likely just a format issue, but hey! it makes it easier on the CPU to transmit.

 You can export using audacity, I do a 48000 sample rate, mono channel. Export settings is a custom ffmpeg export with the container being wav and the codec pcm_u8. 

See if that works