furrtek/portapack-havoc #193: Any new updates

kevtheskin, 2018-07-31:
Hello there,
Any new updates/addons coming

Cheers Kev

RED29, 2018-07-31:
I had the same question... There a lot of receivers that are on hold..

Cheers!
RED

jboone, 2018-07-31:
I understand that @furrtek is busy with other things now. If and when he has time to work on HAVOC, I imagine you will see the updates here in the repository. Please see this comment for more on @furrtek's status: https://github.com/furrtek/portapack-havoc/issues/147#issuecomment-407638044

RED29, 2018-07-31:
@jboone ok thanks. I didn’t know about his comment. Maybe @furrtek is interested in setting up a Patreon account. I would really like to support him montly, and I think a lot other people too!

mwarchut, 2018-08-02:
I'd second the supporting him.. Making it worth his time to move forward..

furrtek, 2018-08-02:
Replied in https://github.com/furrtek/portapack-havoc/issues/147

RED29, 2018-08-05:
Same for me @furrtek thanks for all your work!

unixpunk, 2018-08-07:
+1 Thanks for all your work!

furrtek, 2018-08-07:
Don't make this sound like it's definitive, it's not ! :p
I just started being interested in Zigbee.

mamogodi, 2018-10-10:
Check the book Inside Radio:An attack and defense guide. It deals on zigbee in one of the chapters

closed 2019-03-16 by furrtek