furrtek/portapack-havoc #198: Change time and Date

MaxRomDatastorm, 2018-09-04:
hello
Please someone can suggest me the way for change the time and date ?
thank you in advance

ImDroided, 2018-09-04:
Isn't this done in the settings menu? I'm at work now so I can't check bit
pretty sure that is in there somewhere

furrtek, 2018-09-04:
Hello,
See main menu > Settings > Date/Time

MaxRomDatastorm, 2018-09-05:
Hello, Thank you for the reply
yes I've seen the menù in which there are the possibility to change the data & Time, but the problem is that do not appear any numeric keyboard for insert or change that.
when I highlight a numerical field no numerical grid appear.
Thank you

furrtek, 2018-09-05:
Use the jog wheel when the fields are highlighted.

MaxRomDatastorm, 2018-09-05:
Revolver. Mechanical problem on the round pas. Thanx and Apologize

closed 2018-09-05 by MaxRomDatastorm