furrtek/portapack-havoc #201: Android Application

eudomoniacc, 2018-10-22:
Is there a way to make an android application to use the havoc firmware ?

furrtek, 2018-10-30:
Not sure how what you have in mind.
If your device can be a USB host, you can maybe use a bare HackRF to put/get the raw IQ data.
Mobile OSes (especially Android) might not be reactive enough, though.

eudomoniacc, 2019-03-08:
Well i am back. Since portapack addon is arm based, it is possible for any android device with usb host support like in nethunter etc.. to emulate the interface. Secondly like in Pandwarf if we can manage to communicate hackrf with android device over bluetooth with an extra dongle (firmware must be changed for that :/) that would be very good :) but seems like a dream for now :/