furrtek/portapack-havoc #202: Anything new

kevtheskin, 2018-10-29:
Hello Furrtek, 
How are you. Can you tell us if there will be any new updates before the year is over. I feel that I wasted my money buying the Portapack as I am not a coder . Your firmware is really good and I hope you add some goodies soon. Cheers Kev

eudomoniacc, 2018-10-29:
Well just below i asked about Android platform. So i decided to work on an application which emulates portapack-havoc interface on android to work with the firmware. When it is ready i will be releasing here.

notpike, 2018-10-30:
heh that would be neet.

furrtek, 2018-10-30:
Hello,
> Can you tell us if there will be any new updates before the year is over.
Probably not, most of my time is still being used up by other projects.
> I feel that I wasted my money buying the Portapack as I am not a coder.
You knew what it could do and what it couldn't when you bought it, right ?
> Your firmware is really good and I hope you add some goodies soon.
I hope too :)