furrtek/portapack-havoc #204: Freeze after flashing self compiled firmware

c5e3, 2018-11-30:
I've got a Portapack since around a week now and really like this firmware.
However, I would like to modify some things, so I built the firmware from source, which results in a faulty behaviour. I have to reset the board multiple times to get to boot up. After that, the board completly freezes and no matter how often I reset it, it is not usable at all.
Your release binary or a self-compiled binary from the original firmware work fine though.

c5e3, 2018-11-30:
Apparently commit 609235b broke it for me, since i use a TCXO, that is permanently soldered to the HackRF.

TheCranston, 2018-11-30:
@sharebrained has it fixed in upstream portapack-hackrf.

sgoudelis, 2018-11-30:
@c5e3 sorry to hijack this. I have being looking into getting a TCXO for my hackrfs. Can you share a few pictures about how you did that ?

Thank you

c5e3, 2018-11-30:
@sgoudelis since it is in the metal case now, I don't want to take it out again.
But I simply bought a TCXO addon-board from China, desoldered the TCXO and soldered 3 of its 4 pins to GND, VCC and CLKIN on header P22, accordingly to the layout of the addon board

c5e3, 2018-11-30:
I've merged @sharebrained's commit https://github.com/sharebrained/portapack-hackrf/commit/01bad2805c633b4d53f59a28ec99f3f16e61c41d with the current havoc commit and commeted out some code in portapack.cpp.102:
`auto clkin_status = false; //clock_generator.clkin_status();`

The external CLK is now being recognized and I don't get freezes anymore. Only the UI icon doesn't know anything about the external CLK for now, but this is just a cosmetic fix.

TheCranston, 2018-11-30:
@c5e3 is the merge available?  I'd love to test!  Thanks!

c5e3, 2018-11-30:
I don't want to push this to my fork yet, but here is the binary:
[portapack-h1-havoc.zip](https://github.com/furrtek/portapack-havoc/files/2634136/portapack-h1-havoc.zip)

Don't be confused - the UI will always show external clock for now.

c5e3, 2018-12-02:
update: another issue with @sharebrained's code is, that I have to reset the HackRF after i connect it to power.

TheCranston, 2018-12-02:
@c5e3 so does your HackRF *hang* on initial power up, then after a manual reset it starts functioning?  I remember vaguely from chatting with @sharebrained that his code, and subsequently the @furrtek havoc port, turns off as many of the oscillators as possible and runs from, i think, a single clock.  This was to reduce the clock induced artifacts.  Maybe there is some sort of race condition on cold start that a warm reset side steps...  

For what it's worth, I've not experiencing the cold boot freeze you are reporting.  I'm on the most recent published CPLD and then using the firmware you built a few days ago.

Happy to spend any debug cycles you'd like.  Let me know if you want me to run some specific tests.

c5e3, 2018-12-02:
@TheCranston exactly, it just hangs on initial power up, but I just figured out, that this only happens, when i connect it to my power bank 🤷‍♂️😄

TheCranston, 2018-12-03:
Weird.  I tried my power pack and didn’t have any lockups, but did discover
that in hackrf mode this cut of firmware won’t stream data.  I can see it,
check registers, etc.  but no rf data stream.

c5e3, 2018-12-03:
@TheCranston this is weird indeed. I can transmit and receive over USB without any issues.

TheCranston, 2018-12-03:
I stand corrected...  the portapack firmware (8feb79c71062294c5a83a37b6067f3b2f831034c) Aug 5 commit.  It doesn't stream I/Q data either..  bugger.  I switched back to hackrf proper firmware, working data.  installed sharebrained and it recognized the TCXO and works inside the interface, but switching into hackrf mode, it won't stream.  ugh.  Same behavior for the cut of firmware you built, work's great in the interface, but no stream in hackrf mode.  guess there is something else going on.  I'll raise an issue in the  @sharebraind repo.