furrtek/portapack-havoc #209: please rebuild firmware/portapack-h1-havoc and tag a release

ZeroChaos-, 2019-01-07:
people love binaries and stable releases, pretty please :-)

ablanquart, 2019-01-07:
I believe every build comes with a pre-compiled binary in the /firmware
directory.

Thanks,
adam

ZeroChaos-, 2019-01-08:
perhaps I get confused easily, this one is 7 months old:
https://github.com/furrtek/portapack-havoc/blob/master/firmware/portapack-h1-havoc.bin

which build are you talking about?

unixpunk, 2019-01-08:
That would be it.  DEV is only part-time.  Contributors wanted.

unixpunk, 2019-01-08:
Give this one a shot.

[portapack-h1-havoc.zip](https://github.com/furrtek/portapack-havoc/files/2735296/portapack-h1-havoc.zip)

ZeroChaos-, 2019-01-08:
I am aware, there have been several commits to master in those 7 months and Shmoocon is coming up so I requested an official update.  And no, I don't need an unofficial build, I can build it myself, this request is more for the sake of others.

ryanrohypnol, 2019-01-09:
> Give this one a shot.
> 
> [portapack-h1-havoc.zip](https://github.com/furrtek/portapack-havoc/files/2735296/portapack-h1-havoc.zip)

Is this a current build?

unixpunk, 2019-01-09:
I built it after running a git pull at that time, yes.  It runs on my older portapack without issue.

ryanrohypnol, 2019-01-09:
> I built it after running a git pull at that time, yes. It runs on my older portapack without issue.

Thank you, mine is in the UPS truck right now. So I'm assuming I have a new one... I'll see if your .bin works.

unixpunk, 2019-01-11:
Interesting.  My build works fine on my unit.  You might need to update your CPLD, just guessing since I haven't seen that issue before and I know mine has been updated recently.  HTH

ryanrohypnol, 2019-01-11:
Do you have a Discord / Skype / Email I can contact you on?

unixpunk, 2019-01-11:
Rather keep the discussion here so others can benefit.  Step 2 is what you need, if that's what you were going to ask about: https://github.com/mossmann/hackrf/wiki/Updating-Firmware  Just install or compile from source the hackrf tools, grab that file from here: https://github.com/mossmann/hackrf/raw/master/firmware/cpld/sgpio_if/default.xsvf  The xsvf file is 2 years old, but if you've never updated your hackrf with it yet, then that's the first thing I'd try. 

That's all I can suggest, otherwise you could try flashing the latest hackrf bin first, then the CPLD and then try mine again.  HTH

unixpunk, 2019-01-11:
Oh, also if you go into settings on the working version, ui and disable the splash screen and flash the new image - does it work fine then?  Mine works on or off, but I happen to have it off when I originally tested.

ryanrohypnol, 2019-01-11:
I put it in DFU mode and put the 2017 firmware back on it, then flashed the old Havoc.bin & it's working again. For some reason your build isn't working, is there anything new in the version you compiled?

ryanrohypnol, 2019-01-11:
update: with the old firmware, anytime I use anything it freezes and I have to restart... I'm assuming since I updated the CPLD? This sucks haha :(

ryanrohypnol, 2019-01-11:
He just merged with the fixes for the new portapack, I don't have Ubuntu or I'd build it myself. Does someone mind building it? I'd appreciate the help!

unixpunk, 2019-01-11:
I'm guessing the tx issue with the old bin is because its missing the upstream merge to make the new clock work.  I'm not sure about mine, I had just wiped my build directory, pull, cmake, make firmware, make program without issue, then that's the bin I zipped and attached.

You'll need to build or you'll need someone else to with the new portapack and a fresh build dir to confirm the issue, not sure I can help further on this one.

unixpunk, 2019-01-11:
lol, two minutes

ryanrohypnol, 2019-01-11:
It’s almost 3am here, but I appreciate your help & I’m excited to flash the new firmware in the morning. Thanks man!!

unixpunk, 2019-01-11:
This last pull and build resulted in a brick...Can't even get original portapack dfu blob to boot now, hackrf not found, etc... I'm out.

ryanrohypnol, 2019-01-11:
> This last pull and build resulted in a brick...Can't even get original portapack dfu blob to boot now, hackrf not found, etc... I'm out.

Did you try the portapack windows installer? It has a DFU .bat that worked when mine wasn't being detected!

unixpunk, 2019-01-11:
make program w/dfu is successful as it always has been, just get nothing on the screen but have couple lights.  (not dfu and not hackrf modes, nothing is detected)  Haven't looked at it again yet.

ryanrohypnol, 2019-01-11:
Correct me if I’m wrong but isn’t the DFU feature hardware implemented?

kevtheskin, 2019-01-11:
Hi there, How can you tell which Portapack version you have. I bought mine last year from a seller in Belgium.  Thanks Kev

jboone, 2019-01-12:
@kevtheskin There's an eight digit (YYYYMMDD) revision/date-code printed near the LCD panel and the ShareBrained Technology label.

They're also color-coded. In chronological order, oldest to newest:

- 2014-2016: Green PortaPacks that have the Wolfson audio codec
- 2017: Black PortaPacks with the Asahi-Kasei audio codec
- 2018: Blue PortaPacks with the more precise (<2.5ppm) reference oscillator

ZeroChaos-, 2019-01-12:
aww, now I want a new one :-)

unixpunk, 2019-01-12:
OK, here are two builds.  Older works on my older model, new may work on the newer one, I don't have a new one to test.  Don't fear though, it will never be a brick...  Mine works again now after checking out the previous commit to the latest merge.
[OLDER_portapack-h1-havoc.zip](https://github.com/furrtek/portapack-havoc/files/2752497/OLDER_portapack-h1-havoc.zip)
[NEW_portapack-h1-havoc.zip](https://github.com/furrtek/portapack-havoc/files/2752498/NEW_portapack-h1-havoc.zip)

ryanrohypnol, 2019-01-12:
> OK, here are two builds. Older works on my older green model, new may work on the newer one, I don't have a new one to test. Don't fear though, it will never be a brick... Mine works again now after checking out the previous commit to the latest merge.
> [OLDER_portapack-h1-havoc.zip](https://github.com/furrtek/portapack-havoc/files/2752497/OLDER_portapack-h1-havoc.zip)
> [NEW_portapack-h1-havoc.zip](https://github.com/furrtek/portapack-havoc/files/2752498/NEW_portapack-h1-havoc.zip)

Just tried the new firmware, works well on my new (blue) portapack. The only issue is HackRF mode doesn't work. So I reverted back to the old firmware. Thank you, I'll check back when they iron out the bugs :)

sck-nogas, 2019-01-14:
As another data point, my 20140826 Green Portapak version also did not like this 920b98f commit. I was able to DFU back to stock portapak, then compiled and installed the bbb5dc3 commit to get it functional again. Hopefully, we can find out what is dying on the older models.

jboone, 2019-01-14:
@sck-nogas @unixpunk I'm going to look at this today.

ryanrohypnol, 2019-01-14:
@jboone can you also check to see why the new build is not allowing HackRF mode? I'd appreciate it <3

jboone, 2019-01-14:
@ryanrohypnol This is likely a result of a malformed pull request. I have produced a new pull request #214 that should bring HAVOC up-to-date with the PortaPack code.

@sck-nogas and @unixpunk, if you're so inclined, please give PR #214 a try.

I will also get Travis-CI set up to build binaries.

ryanrohypnol, 2019-01-14:
Made my night! Thank you @jboone

jboone, 2019-01-15:
@furrtek If you are OK giving me push/merge/PR rights to your repo, I can get this stuff merged and tag a release candidate for @ZeroChaos- , who would love to ship a release in Pentoo for Shmoocon, which is in a couple of days. Let me know. :-)

kevtheskin, 2019-01-15:
> @kevtheskin There's an eight digit (YYYYMMDD) revision/date-code printed near the LCD panel and the ShareBrained Technology label.
>
> They're also color-coded. In chronological order, oldest to [...]
Hello Jared. What month last year did the Blue Portapack come out. I bought mine from Maes in Europe last year and its Black?  Cheers Kev

jboone, 2019-01-15:
@kevtheskin The blue PortaPacks came out in the middle of last month -- December 2018.

kevtheskin, 2019-01-15:
> 
> 
> @kevtheskin The blue PortaPacks came out in the middle of last month -- December 2018.

Thanks peeps. K

unixpunk, 2019-01-16:
@jboone I'm not that great with git myself.  I've forked havoc, checked out bbb5dc3, created a 'testing' branch at that HEAD.  Then I added your repo as 'jboone' but failed the git fetch jboone/master with a good list of conflicts.  This is where my git knowledge ends..  If you have guidance I can keep moving, otherwise not had to deal with merge conflicts before..

unixpunk, 2019-01-16:
I gave it a shot with meld using git mergetool and I chose all the REMOTE change merged into the middle pane (not the LOCAL one) but cmake fails now, can't find ui_console.cpp.  Not really sure what I'm doing.  "I'm not a DEV, I just play one on github."  :)

sck-nogas, 2019-01-16:
FYI... I pulled PR 215 and compiled on my Green Portapak and it doesn't crash! According to Git hash I'm running b4ccefa. Seems stable.

jboone, 2019-01-16:
@sck-nogas Thanks for the report. Check HackRF mode, if you can. That seems to be the big issue now.

@unixpunk I appreciate the effort! I believe I've got a workable merge in PR #215 on this repo. I would recommend working from my repo, jboone/portapack-havoc:master until PR #215 gets merged.

beardache, 2019-01-17:
I guess I came to this topic as I've a problem as well with the new 262c030. After compiling for the portapack black version and upgrading the firmware the screen is just blank . Gladly enough I still had the old 441a266d on my machine. That one still works for me.

I don't know what exactly what to do or change to make it work. @jboone Haven't tried your repo yet.

unixpunk, 2019-01-17:
@jboone Just cloned your fork and I have the same issue on make firmware as I reported here: https://github.com/sharebrained/portapack-hackrf/issues/132

EvilDonkey420, 2019-01-18:
> @kevtheskin There's an eight digit (YYYYMMDD) revision/date-code printed near the LCD panel and the ShareBrained Technology label.
>
> They're also color-coded. In chronological order, oldest to [...]

but ofcourse I spent a bunch of money on a black one and then two months later the blue one shows up!!! wtf

furrtek, 2019-02-05:
@jboone I clicked a button and everything worked, what kind of magic was that ?! Thanks !

ZeroChaos-, 2019-02-06:
if you are happy, please make a tag and rebuild the firmware binary in your repo for all to enjoy :-)
oh, and close this ticket

jboone, 2019-02-06:
@furrtek No magic, just steady hard work. But you know how that is... :-)

StephenGenusa, 2019-02-16:
Could someone build and post a zip please? Build instructions are out of date. Thank you.

unixpunk, 2019-02-16:
https://github.com/furrtek/portapack-havoc/pull/215#issuecomment-457451410

StephenGenusa, 2019-02-16:
@unixpunk thank you! That works great.

ZeroChaos-, 2019-03-11:
@furrtek is there something specific holding back rebuilding https://github.com/furrtek/portapack-havoc/blob/master/firmware/portapack-h1-havoc.bin and maybe adding a git tag?  anything I can test for example?

unixpunk, 2019-03-11:
This will get you going, in case you haven't tried it yet: https://github.com/furrtek/portapack-havoc/issues/209#issuecomment-464359619

closed 2020-05-24 by furrtek