furrtek/portapack-havoc #210: Freezing After Transmitting

ryanrohypnol, 2019-01-11:
For some odd reason, when I transmit on the new Portapack, the TX light works but when I press the "back" button pressing STOP. it freezes my device. The only thing I did was update the CPLD and reinstall the same firmware. Prior to updating the CPLD, it was working relatively perfectly. Does anyone have any recommendations or a compiled bin for the new merge? I'd really appreciate the help as I feel like I've tried everything.

Thank you

unixpunk, 2019-01-11:
there's a diff cpld file that's built with make firmware (or make program, can't recall) that you could try.  search the build directory tree for cpld I think.  hth

ryanrohypnol, 2019-01-11:
Where is the build directory tree?

ryanrohypnol, 2019-01-11:
nevermind, I fixed it. I found the default HackRF CPLD. If anyone else is having issues transmitting / freezing please update your CPLD to 

[hackrf_cpld_default.zip](https://github.com/furrtek/portapack-havoc/files/2750564/hackrf_cpld_default.zip)

closed 2019-03-16 by furrtek