furrtek/portapack-havoc #217: Jammer question

ryanrohypnol, 2019-01-20:
I’ve tried to jam 3 devices in my house with no success, one being on 315mhz and another on 302.18mhz. Is there a specific configuration I should be using or is the Jammer just broken? Thank you!

jboone, 2019-01-20:
The CPLD issues we've been discussing may extend to transmit. Once I push the changes (likely Monday afternoon, as all my test hardware is at the office), you should try again.

I should also note that the HackRF has very little transmit/output power -- roughly 10mW or +10dBm. To effectively jam a stronger signal, the HackRF will have to be close to the receiver, and the jamming bandwidth needs to be just wide enough to destroy the target signal. If the bandwidth is too wide, some of your limited transmit power will be wasted on portions of the spectrum where there's no signal to jam.

I'm not familiar with the jamming feature of HAVOC (I'm too upstanding/cowardly to try it), but am guessing there's a way to narrow the jamming bandwidth.

ryanrohypnol, 2019-01-20:
Got it. That makes sense, I will try to move it closer to the receiver rather than the transmitter. I’ve tried jamming on the exact frequencies with no success. Could be like you said, CPLD again. 

Thank you, I’m going to look into ways to increase HackRF’s Tx power.

unixpunk, 2019-01-20:
I think there's specifically an issue with this function, actually.  I also see/hear nothing being received when using this but I can use Microphone and Morse code functions and see/hear the transmits there.

cosmo6110, 2019-03-04:
I tried the jammer function, and it's working for me
But is seems to me the code  is not optimized for power output.
I tried the yardstick and (arduino and cc1101) with ask/ook to jam my car (2fsk).
 I can say it's very big different's in power output.

cosmo6110, 2019-03-04:
I compiled this code.
https://github.com/Lefinnois/hackrf_ook
and jammed my nexa remote
./hackrf_ook -s 0 -b 1700 -0 1284 -1 416 -p 10000 -m 1110101010111010101010101 -f 433920000 -g
It's working good.

VVaReD, 2019-12-10:
I tested it works on the wifi but not on networks gsm ("Free" french gsm network) :-(

furrtek, 2019-12-10:
-Frequency range
-Power
-Phones switching bands depending on availability

Not a firmware issue.

VVaReD, 2019-12-10:
thank you for responding so quickly Furrtek ;-)