furrtek/portapack-havoc #218: build issue / make firmware

DRAHCORE, 2019-01-21:
'cmake ..' works fine,
But, when I run 'make firmware', the issue is next : 

[100%] Generating portapack-h1-havoc.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
make[3]: *** [firmware/CMakeFiles/program.dir/build.make:57: firmware/CMakeFiles/program] Error 74
make[2]: *** [CMakeFiles/Makefile2:164: firmware/CMakeFiles/program.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:171: firmware/CMakeFiles/program.dir/rule] Error 2
make: *** [Makefile:144: program] Error 2

This issue is when my Portapack is connected with the Hackrf mode, without this mode the issue was :

[100%] Generating portapack-h1-havoc.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        22056 bytes
Download done.
dfu-util: unable to read DFU status after completion
File size 1048576 bytes.
hackrf_open() failed: HackRF not found (-5)
make[3]: *** [firmware/CMakeFiles/program.dir/build.make:59: firmware/CMakeFiles/program] Error 1
make[2]: *** [CMakeFiles/Makefile2:164: firmware/CMakeFiles/program.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:171: firmware/CMakeFiles/program.dir/rule] Error 2
make: *** [Makefile:144: program] Error 2

Did I install a wrong something? I'm a frenchy novice!

jboone, 2019-01-22:
@DRAHCORE In the first example, do you have the HackRF in DFU mode? DFU mode is when you connect the HackRF via USB while holding down the blue "DFU" button. There are two blue buttons. The DFU button is the one that is closer to the antenna.

In the second example, it looks like you *are* in DFU mode. However, the make file is trying to program the HackRF too soon after loading the DFU code. There should be a sleep of three seconds or so between the DFU step and the HackRF flash programming step.

You might want to try my fork of the HAVOC repo, which has these fixes incorporated. However, I am working on another problem at this time, so it still might not work for you. I should have everything fixed tomorrow. I will come back to this thread and let you know when it is done.

DRAHCORE, 2019-02-13:
Finaly, i don’t know why, but it’s running...

closed 2019-03-16 by furrtek