furrtek/portapack-havoc #222: HackRF Mode not functioning

PPH12, 2019-02-10:
Switching to HackRF mode appears to not work. Blank screen on PortaPack, USB led is off and HackRF is not presented as USB device to host machine.

F0GPI, 2019-02-10:
hi, have you checked your usb cable?

gadjodilo83, 2019-02-11:
I have a similar problem. if I choose hackrf mode, then the radio station are very disgustingly. the voices are very high ... like mickey mouse...

i-am-tof, 2019-02-17:
Hi,
On my PortaPack 20181029, the only correct firmware while in HackRF mode is git dc5d6fe.
With all other versions including last one from source, PortaPack is still working in HackRF mode, but the screen is always switched on printing a blue screen with electronic symbols in white.
I'm not sure which XSVF file is the good one to be written to CPLD. It does the same with Jboone's https://github.com/jboone/hackrf/raw/master/firmware/cpld/sgpio_if/default.xsvf

jboone, 2019-02-19:
@i-am-tof That screen is a new feature! It displays the state of the radio when operating it in HackRF mode with a PC. Maybe I need to enhance it to make it look less mysterious and blank at start-up.

jboone, 2019-02-19:
@gadjodilo83 There was an issue addressed in early January that sounds like what you describe. Please try the latest build of firmware and programming the latest CPLD file onto your HackRF.

i-am-tof, 2019-02-19:
**@jboone** Thanks for your answer and for your work, that's a great idea but...
1) when connected during hours to a PC, it will still use its onboard screen, so there should be a setting to allow it to be shutted off, avoiding this feature to damage the PortaPack's screen with time.
2) On a laptop that will use more power from batteries than before, so we should have the choice.

furrtek, 2019-02-19:
I'm not aware of modern LCDs suffering from burn-in but having the choice to just turn off the backlight would be nice.

i-am-tof, 2019-02-19:
**@jboone** An idea : use the screensaver setting, so we just have to push a button to see this feature, then go back to the screensaver when selected delay is elapsed.

jboone, 2019-03-21:
@i-am-tof I would not use a HAVOC screensaver setting, as some people do not use the HAVOC PortaPack firmware and would not have access to the setting. I have added a feature-issue to the HackRF project to address this via a host tool argument: https://github.com/mossmann/hackrf/issues/608

@furrtek please feel free to close this issue.

closed 2019-11-01 by furrtek

nocomp, 2019-11-15:
hi folks,
first of all, thank you so much for this great firmware, just dreaming awake! future features look awesome!
i ve flashed last version, havoc mode is awesome, i use it with a powerbank, but when i activate hack rf mode, screen is all black.
am i missing something to do?
thxx for your help