furrtek/portapack-havoc #224: integrate the latest HackRF firmware 2018.01.1

aneitzel, 2019-02-14:
Hello,

can you integrate the latest HackRF firmware 2018.01.1 (API:1.02) into Havoc binary on GHub please.

I flashed https://github.com/furrtek/portapack-havoc/blob/master/firmware/portapack-h1-havoc.bin to my hackRF and noticed, that it includes the old Firmware Version: 2017.02.1

Thank you!

unixpunk, 2019-02-14:
Here is one I built not too long ago that should be what you need: https://github.com/furrtek/portapack-havoc/pull/215#issuecomment-457451410

xerospawn, 2019-02-14:
> (comment)

Is there anyone doing these tweaks and updates on Youtube? I dont have much
time for mine, but I really enjoy following along with what you all are
doing- would be awesome if it were in video. What's working, not working,
etc.

aneitzel, 2019-02-15:
hey unixpunkt, thank you! Now it shows Firmware as gitbeexxx

is this korrekt? ;-) ?? See below

Andres-iMac:jboone-havoc alpha$ hackrf_info 
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f391ee1
Board ID Number: 2 (HackRF One)
Firmware Version: **git-e433bee (API:1.03)**
Part ID Number: 0xa000cb3c 0x00784758
Andres-iMac:jboone-havoc alpha$

unixpunk, 2019-02-15:
Yes, it was built from jboone's fork after working out some compatibility with the older portapack models and some cmake changes.  I believe those changes have been merged here now, but I have not verified or tried to build from here again yet.

vicdzen, 2019-05-07:
I have tested binary mentioned in this ticket. Portapack mode works fine as expected, but while using in HackRF mode, in Win 10, all SDR soft ( SDRAngel, SDR# etc output pitched up and juggy sound while deconding audio (FM stations). Seems like something with sound sample rate. Anyone have such an issue?

jboone, 2019-05-07:
@vicdzen The binaries mentioned in this thread are out of date. The issue described did exist in an old firmware version. I would recommend building from the latest source. Or, here is a more recent build you may try: https://portapack-h1-builds.s3.amazonaws.com/jboone/portapack-havoc/12/12.1/build/firmware/portapack-h1-havoc-1352e16.zip

aneitzel, 2019-11-16:
Hey folks, something is wrong with the included HACKRF mode firmware in jboone build. SDRangel dont work with this version - i got always error then try to start sample source in SDRangel. Flashed back my portapack to HackRF original firmware 2018.01.1 (API:1.02) works. 

I noticed that there is a API version changed also in compare to 2018.01.1, see below:

HackRF info with original 2018.01.1 (API:1.02)
d:\HackRF\HackRF_tools>hackrf_info.exe
hackrf_info version: git-9bbbbbf
libhackrf version: git-9bbbbbf (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f391ee1
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00784758

HackRF info with jboones build
d:\HackRF\HackRF_tools>hackrf_info.exe
hackrf_info version: git-9bbbbbf
libhackrf version: git-9bbbbbf (0.5)
Found HackRF
Index: 0
Board ID Number: 2 (HackRF One)
Firmware Version: git-eae870d (API:1.03)
Part ID Number: 0xa000cb3c 0x00784758

seems it is not the original hackRF fw :-(

aneitzel, 2019-11-16:
pse note that with jboones build the serial number output is missing - iam not sure if this is the source why SDRangel cant deal with hackRF under this fw.

jboone, 2019-11-16:
Please see issue #251, which I believe will address your issue. Try the latest binary from here: https://jboone.github.io/portapack-havoc-nightly/