furrtek/portapack-havoc #226: I updated the firmware and CPLD, all successfully ,but the screen is always blank, dark no light.

freeworld1981, 2019-02-27:
I updated the firmware and CPLD, all successfully ,but the screen is always blank, dark no light.

the led only green/yellow/red light.

where is the problem?

pipi111, 2019-03-15:
same problem

alphaomicron, 2019-03-21:
After updating, the play dead mode may be enabled; this is indicated by a random firmware error, such as the following:

Firmware error 0x14008000

To exit this, press: up - down - left - right - center button. You can disable this by editing: settings -> disable login. The setting will be saved for next time.

alphaomicron, 2019-03-21:
And also RTFM.

kevtheskin, 2019-04-26:
Hello there , I am getting the same error. And the exit suggestion does not work? Kev

yadox666, 2019-07-20:
Same, no fake error code on screen. But my problem does not appear in an original hackrf one and original portapack. In the original kit it works like a charm. But a friend bought a hackrf one clone in aliexpress branded as hackrf one 2014. It's not a problem of the portapack, because when I switched to my original hackrf it worked from the beginning. 

Flashing portpack original software works, but when flashing portapack havoc bin it does not boot, it shows black screen, no logo, and in dmesg it does not appear any new device. I tried to reflash dfu, calibration data and spi flash without success. So there must be some chinese cloned hackrf boards with some problems. Maybe spi size? No more ideas to test. Thank you and congratulations for your development and work.

hkn300, 2019-09-24:
i had same problem , some how latest build not working (i got latest Chinese colones.). i am just getting black screen. but got luck older build  works for me, but functionality not great. like you said after many many  unsuccessful boot finally i was able to see menus , after reset just black screen. i am currently stuck with older version . had some performances issue

closed 2020-05-24 by furrtek