furrtek/portapack-havoc #228: Generating hackrf.bin, hackrf.img Fails

gadjodilo83, 2019-03-11:
dear havoc community :)

is it possible that at present no image can be created from the source code?

i have the following error:

Traceback (most recent call last):
  File "../../../firmware/tools/make_image_chunk.py", line 62, in <module>
    raise RuntimeError('image size of %d exceeds device size of %d bytes' % (len(input_image), input_image_max_length))
RuntimeError: image size of 34308 exceeds device size of 32768 bytes
firmware/baseband/CMakeFiles/baseband.dir/build.make:338: recipe for target 'firmware/baseband/hackrf.bin' failed
make[3]: *** [firmware/baseband/hackrf.bin] Error 1
make[3]: *** Deleting file 'firmware/baseband/hackrf.bin'
CMakeFiles/Makefile2:417: recipe for target 'firmware/baseband/CMakeFiles/baseband.dir/all' failed
make[2]: *** [firmware/baseband/CMakeFiles/baseband.dir/all] Error 2
CMakeFiles/Makefile2:138: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
Makefile:131: recipe for target 'firmware' failed
make: *** [firmware] Error 2

thanks for any replay
gadjodilo83 :)

jboone, 2019-03-11:
@gadjodilo83 You're correct. That is an error caused by the hackrf master branch building a binary that's larger than 32K. There's a Python tool that enforces that criteria. However, it only applies to baseband firmware images, not the HackRF binary. There is a commit on jboone/hackrf that fixes the issue. I will merge these changes into jboone/portapack-havoc, and you can try building from that. I will update this issue when it's ready.

gadjodilo83, 2019-03-11:
hello @jboone

Thanks for your quick reply! I will try again later, maybe in the next days or so :) Thank you for your help with this project and all those who keep the project alive. 

gadjodilo83

jboone, 2019-03-11:
@gadjodilo83 Please try building from jboone/portapack-havoc now.

I have a Travis-CI build showing that it builds successfully, and I tried the build artifacts from https://portapack-h1-builds.s3.amazonaws.com/jboone/portapack-havoc/12/12.1/build/firmware/portapack-h1-havoc-1352e16.tar.bz2 and they work. I've submitted PR #229 to address this.

Please close this issue if building works for you.

closed 2019-03-11 by gadjodilo83

unixpunk, 2019-03-12:
Latest pull from jboone/portapack-havoc compiled for me.  Thanks!