furrtek/portapack-havoc #230: What is LGE Tool included in the new firmware?

decadenzio, 2019-03-12:
I just can't find the info in the wiki or somewhere else.
I found this page on wikipedia about Laser Guided Energy, kind of a new weaponry system?

https://en.wikipedia.org/wiki/Electrolaser

furrtek, 2019-03-12:
LGE is a chain of laser tag centers in France.

closed 2019-03-16 by furrtek

xvostikk, 2019-08-16:
> LGE is a chain of laser tag centers in France.

Can you explain more? I am testing now the HackRF with Portapack and it is really difficult to understand how i can use this Transmitter.
Thank you in advance.

furrtek, 2019-08-16:
It's completely useless if you're not located in or near an LGE laser tag center.
If by any luck you are, then you get pretty much do what you want: spoof hits, change player and team names, start and terminate the session...

xvostikk, 2019-08-16:
That's why the description of the tool is really necessary. For example in Germany we have not these centers but we use HAVOC. ;)

furrtek, 2019-08-16:
I mentionned that in the previous reply.