furrtek/portapack-havoc #232: Problem when using with sdr software

alphaomicron, 2019-03-21:
When Portapack is put into Hackrf mode in order to use it with sdr sharp or SDR console v3 I don't get any signal waterfall. Also same happens if I use grgsm_scanner I don't get any signals. Any pointers? 
Firmware Version: 2017.02.1 (API:1.02)

alphaomicron, 2019-03-21:
Upgraded firmware to Firmware Version: git-eae870d (API:1.03)
Still same issues.

cosmo6110, 2019-03-21:
Have you install the  http://zadig.akeo.ie/ 
drivers for windows?
http://recolog.blogspot.com/2018/02/installing-drivers-for-rtl-sdr-and.html

jboone, 2019-03-21:
@alphaomicron First, what operating system and version are you using? What is the state of the LEDs while you are attempting to stream? What is the output from hackrf_info, and can you do a command-line hackrf_transfer -r (receive) successfully?

If we can't uncover anything with that information, can you try my latest HackRF build? Flash it directly on your device, and verify that it works stand-alone (not from PortaPack/HAVOC):

https://github.com/jboone/hackrf-nightly/raw/master/HackRF-2019-03-21-6e5b31d.tar.xz

If that doesn't work, try the last released HackRF firmware and CPLD bitstream:

https://github.com/mossmann/hackrf/releases/tag/v2018.01.1

We need to tease apart if there's a hardware or host operating system issue, or if it's something in the hand-off from PortaPack/HAVOC to HackRF firmware.

alphaomicron, 2019-03-21:
@jboone Sorry for not following the right procedure. Running on a host with win10 and guest is latest Kali. State of Leds: 3.3 Volt light on (green), Radio frequency light on (red), usb light on (green). All the rest of the leds are off. Also when I change the frequency in SDR console or sdrsharp th same frequency shows up in the new blue screen of portapack.

`hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Board ID Number: 2 (HackRF One)
Firmware Version: git-eae870d (API:1.03)
Part ID Number: 0xa000cb3c 0x00534f57
`

hackrf_trasnfer -r
`hackrf_transfer -s 2 -f 451000000 -r test
call hackrf_set_sample_rate(2 Hz/0.000 MHz)
call hackrf_set_freq(451000000 Hz/451.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 2.124 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 2.12370 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
`

jboone, 2019-03-21:
@alphaomicron That's helpful, thanks. You're specifying a sampling rate of 2 Hz, which I'm sure is not what you intend. Only sample rates between 8 MHz and 20 MHz are officially supported by the HackRF. Please try "-s 8000000" instead and let me know if it works better.

alphaomicron, 2019-03-21:
`hackrf_transfer -s 8000000 -f 451000000 -r test
call hackrf_set_sample_rate(8000000 Hz/8.000 MHz)
call hackrf_set_freq(451000000 Hz/451.000 MHz)
Stop with Ctrl-C
 2.6 MiB / 1.001 sec =  2.6 MiB/second
 0.0 MiB / 1.001 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 2.00178 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
`

jboone, 2019-03-21:
@alphaomicron Can you try the last released HackRF firmware and CPLD bitstream? 

https://github.com/mossmann/hackrf/releases/tag/v2018.01.1

If it still doesn't work, can you try this via the host operating system, or reboot into Kali and test? Running high-bandwidth USB streaming into a virtual machine has been problematic in some cases.

alphaomicron, 2019-03-21:
@jboone  Ok Im so banging my head on the wall!
`hackrf_transfer -s 8000000 -f 451000000 -r test
call hackrf_set_sample_rate(8000000 Hz/8.000 MHz)
call hackrf_set_freq(451000000 Hz/451.000 MHz)
Stop with Ctrl-C
16.5 MiB / 1.001 sec = 16.5 MiB/second
16.8 MiB / 1.000 sec = 16.8 MiB/second
16.5 MiB / 1.001 sec = 16.5 MiB/second
16.8 MiB / 1.001 sec = 16.8 MiB/second
16.8 MiB / 1.000 sec = 16.8 MiB/second
16.5 MiB / 1.001 sec = 16.5 MiB/second
16.8 MiB / 1.001 sec = 16.8 MiB/second
16.5 MiB / 1.001 sec = 16.5 MiB/second
^CCaught signal 2
 4.2 MiB / 0.260 sec = 16.1 MiB/second

Exiting...
Total time: 8.26623 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
`
Also now receiving normally in SDR console and sdrsharp...I tried another data cable and it worked. Needless to say I had tried several another before posting but not this one.  No need to try another firmware I guess :)

alphaomicron, 2019-03-21:
Thank you for your help @jboone

closed 2019-03-21 by alphaomicron