furrtek/portapack-havoc #233: adsb_map.py=> TypeError: can only concatenate str (not "bytes") to str

r1k0, 2019-04-03:
```
➜  tools git:(master) ✗ pwd
/home/rik/portapack-havoc/firmware/tools
➜  tools git:(master) ✗ ll ../../sdcard/ADSB 
total 474M
-rw-r--r-- 1 rik rik  90K Mar 31 12:41 airlines.db
-rw-r--r-- 1 rik rik  29M Apr  3 17:22 world_map.jpg
➜  tools git:(master) ./adsb_map.py 
Traceback (most recent call last):
  File "./adsb_map.py", line 42, in <module>
    line += struct.pack('H', pixel_lcd)
TypeError: can only concatenate str (not "bytes") to str
➜  tools git:(master) python --version
Python 3.7.2
➜  tools git:(master) vim adsb_map.py 
➜  tools git:(master) ✗ ./adsb_map.py  
➜  tools git:(master) ✗ ll ../../sdcard/ADSB 
total 474M
-rw-r--r-- 1 rik rik  90K Mar 31 12:41 airlines.db
-rw-r--r-- 1 rik rik 445M Apr  3 17:39 world_map.bin
-rw-r--r-- 1 rik rik  29M Apr  3 17:22 world_map.jpg
➜  tools git:(master) ✗ git diff adsb_map.py | cat
diff --git a/firmware/tools/adsb_map.py b/firmware/tools/adsb_map.py
index bb9b71b0..e279def4 100755
--- a/firmware/tools/adsb_map.py
+++ b/firmware/tools/adsb_map.py
@@ -34,7 +34,7 @@ outfile.write(struct.pack('H', im.size[0]))
 outfile.write(struct.pack('H', im.size[1]))
 
 for y in range (0, im.size[1]):
-	line = ''
+	line = b''
 	for x in range (0, im.size[0]):
 		pixel_lcd = (pix[x, y][0] >> 3) << 11
 		pixel_lcd |= (pix[x, y][1] >> 2) << 5
➜  tools git:(master) ✗ 

```
map seems to load fine.

midnyteshade, 2019-04-13:
Python 3.x doesn't have implicit conversion\encoding for bytes to string.  If you use python 2.x it will work fine.

r1k0, 2019-04-13:
just thought I'd log it here but it's ok to close

closed 2019-04-13 by r1k0

kedare, 2019-06-28:
Any hint on how to make it work with Python 3 ? As PIL is a PITA to install on Python 2 (vs Pillow)

I tried to make the following changes, the file is generated without errors but the rendering doesn't look like a map (It shows something on the screen however)

I'm not used to binary file manipulation using Python.

```
diff --git a/firmware/tools/adsb_map.py b/firmware/tools/adsb_map.py
index bb9b71b0..e5b1a2dd 100755
--- a/firmware/tools/adsb_map.py
+++ b/firmware/tools/adsb_map.py
@@ -39,6 +39,6 @@ for y in range (0, im.size[1]):
                pixel_lcd = (pix[x, y][0] >> 3) << 11
                pixel_lcd |= (pix[x, y][1] >> 2) << 5
                pixel_lcd |= (pix[x, y][2] >> 3)
-               line += struct.pack('H', pixel_lcd)
-       outfile.write(line)
+               line += str(struct.pack('H', pixel_lcd))
+       outfile.write(line.encode())
        print(str(y) + '/' + str(im.size[1]) + '\r', end="")
```

r1k0, 2019-06-29:
the diff in my OP solved it for me.